mne-tools/mne-python #5455: Make "raw" BrainVision events accessible

cbrnr, 2018-08-20:
Currently, markers stored in BrainVision files are first converted to a stim channel, which then needs to be parsed with `mne.find_events` to extract events. This is error-prone and does not work for arbitrary marker types. #5387 made this a little more generic, but I have markers that do not follow the `Cxxx` style (where `C` is a character and `xxx` is a number), which is typical for artifact annotations. Here are some example markers I have:

```
Mk1=New Segment,,1,1,0,20110228132350000000
Mk2=Comment,Start Epoch,1,1,0
Mk3=Comment,DC mode,1,1,0
Mk4=Comment,CMS in range,1,1,0
Mk5=Comment,Imp. Check,1,1,0
Mk6=Bad Interval,Bad Min-Max,119,125,0
Mk7=Bad Interval,Bad Min-Max,531,144,0
Mk8=Bad Interval,Bad Min-Max,533,144,0
Mk9=Bad Interval,Bad Min-Max,534,141,0
Mk10=Bad Interval,Multiple,968,178,0
Mk11=Bad Interval,Bad Min-Max,968,129,0
Mk12=Bad Interval,Multiple,970,162,0
Mk13=Bad Interval,Multiple,970,162,0
Mk14=Bad Interval,Multiple,986,166,0
Mk15=Bad Interval,Bad Min-Max,989,163,0
Mk16=Bad Interval,Bad Min-Max,993,171,0
Mk17=Bad Interval,Bad Min-Max,996,151,0
Mk18=Bad Interval,Multiple,1318,138,0
Mk19=Stimulus,S 11,2599,1,0
Mk20=Stimulus,S120,3383,1,0
Mk21=Stimulus,S  2,4229,1,0
Mk22=Stimulus,S 11,4622,1,0
Mk23=Stimulus,S 90,5395,1,0
```

I think it would be great to have a method that returns the raw marker structure, similar to what is already available for EDF files (`raw.find_edf_events`). This method could either be called `raw.find_brainvision_events`, or we could add a generic `raw.find_raw_events` (better name suggestions welcome) method that works for arbitrary file types. Personally, I'd prefer the latter because then I don't have to remember which file type I originally loaded.

On a related note, I've tried to load files with this marker structure using a combination of the `trig_shift_by_type` and `event_id` arguments, but it turns out that some events get lost in the process. Maybe I did something wrong, but it seems like this should also be fixed (in a separate PR).

agramfort, 2018-08-20:
that's exactly what the plan is and doing it via an Annotations object.
the objective if to expose the markers in raw.annotations

cbrnr, 2018-08-20:
Is anyone working on this plan? Is there a related discussion, PR, or issue? Sorry, I had this in the back of my head but I can't remember what the current status is. If we want a quick fix, we might consider adding what I wrote in #5456 for the time being.

agramfort, 2018-08-20:
you should coordinate with @massich for this.

he's busy improving our support of Annotations

massich, 2018-08-21:
This is the same behavior as Annotations. So we should integrate them together

cbrnr, 2018-08-21:
Let me know how I can help.

cbrnr, 2018-10-04:
Will be resolved in #5460.

closed 2018-10-04 by cbrnr