mne-tools/mne-python #5457: interactive volume viz

jasmainak, 2018-08-20:
I'm raising a new issue to keep track of things that need to be done but are not part of https://github.com/mne-tools/mne-python/pull/5329

1. I think we should have an entry in the visualization section in http://martinos.org/mne/stable/documentation.html on source estimates. Basically showing every option to visualize source estimation results : on surface, on volume, dipoles.
2. Visualization of vector fields on top of functional maps (after @britta-wstnr implements the computation aspect)
3. Ability to save an animation of the functional map
4. Getting a max: temporal / spatial (globally or within x mm of mouse click) / spatiotemporal

cc @agramfort @sarangnemo

agramfort, 2018-08-21:
great summary.

I think you can start with 1 starting from current master. Then add to it
the interactive volume viz
when merged.

larsoner, 2018-09-01:
I added a point that these plotters can be extended to work with volume data using the ribbon definition, hopefully easy to copy paste over from my closed glass brain PR

larsoner, 2018-09-01:
Argh I mean with surface data

mmagnuski, 2018-10-09:
I've ticked off viz tutorial cause it's done. :)

larsoner, 2019-04-16:
@jasmainak marking this for the sprint in case you want to work on it

jasmainak, 2019-04-16:
okay, perfect!

larsoner, 2021-02-19:
I think most of this is YAGNI at this point and/or can be obtained from the 3D viz code so I'll close. But let's reopen if people are still enthusiastic about some of these features

closed 2021-02-19 by larsoner