mne-tools/mne-python #5463: mlab rendering issue

jhouck, 2018-08-21:
I have a stock version of mne-python 0.16.2 recently installed via the updated environment.yml.  In the mne environment the gui will open but no images render -- e.g. `mne coreg` gives me the coreg window with an empty frame where the 3D display should be.  Coregistration seems to be doing something when I push buttons, and I can set options etc., but no visualization appears on-screen.  The console fills with VTK errors that seem to deal with vtkShaderProgram.  This is using the sample data.

Similarly, mlab figures don't render.  It's not that they're non-interactive -- the window frame appears but no image renders.  Saving produces a solid black png file.  There is no error, but also no display.  This is using the sample data with sphx-glr-auto-examples-visualization-plot-eeg-on-scalp-py; other examples like sphx-glr-auto-examples-visualization-plot-xhemi-py throw the vtkShaderProgram error and then crash with a seg fault.

If I set ETS_TOOLKIT I get the error "No pyface.toolkits plugin could be loaded for qt4" on attempting to import mlab.  I use a network install for anaconda and encounter the same error across multiple machines. There is no matplotlibrc file.

These are things that have worked recently on my local machine under slightly older versions.  Is there an obvious fix here or should I revert to an older snapshot of my anaconda setup?
 
glxinfo on my local machine tells me:
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL version string: 2.1 Mesa 11.0.7
OpenGL shading language version string: 1.30
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)

mne.sys_info says:
Platform:      Linux-2.6.32-696.18.7.el6.x86_64-x86_64-with-centos-6.9-Final
Python:        3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 17:14:51)  [GCC 7.2.0]
Executable:    /export/research/analysis/human/jhouck/shared/tools/anaconda/envs/mne/bin/python
CPU:           x86_64: 8 cores
Memory:        5.8 GB

mne:           0.16.2
numpy:         1.15.0 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=Qt5Agg}

sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.6.1 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.4

Thanks!

larsoner, 2018-08-21:
Your OpenGL version is too old. Can you try what @nugenta suggested in https://github.com/mne-tools/mne-python/issues/5390#issuecomment-411817218 ?

jhouck, 2018-08-21:
Well, that worked.  Thanks!  Anything that keeps the old hardware usable is much appreciated.

larsoner, 2018-08-21:
Glad it worked -- please also check that these instructions (in `master` a couple of hours ago) would have helped

https://github.com/mne-tools/mne-python/pull/5454/files#diff-d309036842deb71073e3debe10ab497fR153

closed 2018-08-21 by larsoner