mne-tools/mne-python #5476: report won't add plot_alignment figure

dnacombo, 2018-08-28:
Running the code below should open a plot_alignment figure, add it to a report, save the report and close.
It only works line by line (in spyder, F9 runs current line of the editor), but not when running the whole file, be it with "play" button at the top of my editor (spyder again), or with `python test.py` in a linux terminal.

```Python
from mne.io import read_raw_fif
from mne.datasets import sample
from mne.report import Report
from mne.viz import plot_alignment

raw = read_raw_fif(sample.data_path() +
                             '/MEG/sample/sample_audvis_raw.fif')

report = Report(title='Checking brain-MEG coregistration')

meg = ['helmet', 'sensors', 'ref']
fig = plot_alignment(raw.info, trans=None, dig=False, eeg=False,
                     surfaces=[], meg=meg, coord_frame='meg',
                     verbose=True)

report.add_figs_to_section(fig,captions='test')

report.save('report_test.html', overwrite=True, open_browser=True) 

```

#### Actual results
in a terminal:
```
python test.py
Opening raw data file //MNE-sample-data/MEG/sample/sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Current compensation grade : 0
Embedding : jquery-1.10.2.min.js
Embedding : jquery-ui.min.js
Embedding : bootstrap.min.js
Embedding : jquery-ui.min.css
Embedding : bootstrap.min.css
Getting helmet for system 306m
Segmentation fault (core dumped)

```
in spyder's ipython console:
```
Opening raw data file /network/lustre/iss01/cenir/analyse/meeg/CAUSALITY_JUELICH/Collab_Juelich_CTPS+Causality_Data/Analysis_Max/MNE_analysis/data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Current compensation grade : 0
Embedding : jquery-1.10.2.min.js
Embedding : jquery-ui.min.js
Embedding : bootstrap.min.js
Embedding : jquery-ui.min.css
Embedding : bootstrap.min.css
Getting helmet for system 306m
Traceback (most recent call last):

  File "<ipython-input-98-d30dad6db3c2>", line 1, in <module>
    runfile('/home/maximilien.chaumon/owncloud/Lab/Projects/CTPS/code/test.py', wdir='/home/maximilien.chaumon/owncloud/Lab/Projects/CTPS/code')

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/spyder_kernels/customize/spydercustomize.py", line 678, in runfile
    execfile(filename, namespace)

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/spyder_kernels/customize/spydercustomize.py", line 106, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/maximilien.chaumon/owncloud/Lab/Projects/CTPS/code/test.py", line 26, in <module>
    report.add_figs_to_section(fig,captions='test')

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 967, in add_figs_to_section
    comments=comments)

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 922, in _add_figs_to_section
    img = _fig_to_img(fig, image_format, scale)

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 77, in _fig_to_img
    img = mlab.screenshot(figure=fig)

  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 352, in screenshot
    out.shape = shape

ValueError: cannot reshape array of size 12 into shape (0,0,3)

```

#### Additional information
Paste the output of `mne.sys_info()` here.
```
Platform:      Linux-4.15.0-32-generic-x86_64-with-debian-buster-sid
Python:        3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 17:14:51)  [GCC 7.2.0]
Executable:    /home/maximilien.chaumon/anaconda3/envs/mne/bin/python
CPU:           x86_64: 8 cores
Memory:        31.3 GB

mne:           0.17.dev0
numpy:         1.14.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=Qt5Agg}

sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.6.1 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.2
```

larsoner, 2018-08-28:
Sometimes Spyder does not get along so well with Mayavi when re-running scripts. Do you get this error even if you close the terminal, then run with IT (which should open a fresh terminal)?

dnacombo, 2018-08-30:
Don't know what IT stands for here.
Running the script completely outside of spyder in a terminal returns what is shown in the second code block above (segmentation fault).

larsoner, 2018-09-04:
Sorry "IT" was an autocorrect typo. I meant F5 :)

Also Spyder has a backend option for MPL -- you could try setting this to `PyQt5` and it might work better.

I'm not sure why you would get a seg fault when running. Can you try enabling faulthandler in your script with:
```
import faulthandler
faulthandler.enable()
```

dnacombo, 2018-09-04:
with faulthandler enabled:

```
Fatal Python error: Segmentation fault

Current thread 0x00007efed80fb740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 569 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 77 in _fig_to_img
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 922 in _add_figs_to_section
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 967 in add_figs_to_section
  File "test.py", line 27 in <module>
Segmentation fault (core dumped)

```
Also changing graphics backend of spyder's ipython console to PyQt5 does not solve the issue.

larsoner, 2018-09-04:
Argh. This seems like a VTK bug :( Some ideas for how to proceed (sorry for the hassle but debugging these things is a pain):

1. Just for completeness, what are the outputs of:
    ```
    $ glxinfo | grep renderer
    $ glxinfo | grep version
    ```
2. To try to make a minimal example to open for Mayavi, does this also error/work/segfault for Spyder-F5/Spyder-line-by-line/console, respectively:
    ```
    from mayavi import mlab
    mlab.test_plot3d()
    mlab.screenshot()
    ```
3. Can you try adding `import matplotlib.pyplot as plt; assert plt.get_backend() == 'Qt5Agg'` at the top of your file and try running it again and see if it still errors/works/segfaults, respectively? It might help forcing some library loads.

dnacombo, 2018-09-04:
1.
```
$ glxinfo | grep renderer
OpenGL renderer string: Quadro K2200/PCIe/SSE2
$ glxinfo | grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 390.48
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 390.48
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 390.48
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 
```
2.
the 3 lines run fine line by line in spyder. (notice warning about Qt4)

```
In [1]: from mayavi import mlab
Changing backend to Qt4 for Mayavi

In [2]: mlab.test_plot3d()
Out[2]: <mayavi.modules.surface.Surface at 0x7fefa0446570>

In [3]: mlab.screenshot()
Out[3]: 
array([[[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]],

       [[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]],

       [[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]],

       ...,

       [[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]],

       [[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]],

       [[127, 127, 127],
        [127, 127, 127],
        [127, 127, 127],
        ...,
        [127, 127, 127],
        [127, 127, 127],
        [127, 127, 127]]], dtype=uint8)

In [4]: 
```
with F5 too (no output)
from the terminal (faulthandler enabled):
```
$ python untitled0.py 
Fatal Python error: Segmentation fault

Current thread 0x00007f6b1feaa740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 569 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "untitled0.py", line 13 in <module>
Segmentation fault (core dumped)
```
3. adding import matplotlib.pyplot as plt; assert plt.get_backend() == 'Qt5Agg' in this test script
```
$ python untitled0.py 
Fatal Python error: Segmentation fault

Current thread 0x00007f3af28fa740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 569 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "untitled0.py", line 13 in <module>
Segmentation fault (core dumped)
```

in my script

```
Fatal Python error: Segmentation fault

Current thread 0x00007fbc11807740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 569 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 77 in _fig_to_img
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 922 in _add_figs_to_section
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 967 in add_figs_to_section
  File "test.py", line 29 in <module>
Segmentation fault (core dumped)
```

larsoner, 2018-09-04:
Okay one more idea. From Spyder and terminal, do:
```
>>> import vtk
>>> vtk
```
To tell us where vtk is being imported. Then finally:
```
$ conda list
```
To get your environment. If two vtk libraries are listed, do:
```
$ conda uninstall vtk
$ pip uninstall mayavi vtk
$ pip install vtk
$ pip install mayavi
```
Then see if anything works better. We might as well test the minimal example code with test_plot3d because it's simplest.

dnacombo, 2018-09-04:
in spyder:
```
<module 'vtk' from '/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/vtk/__init__.py'>
```
indeed two vtk listed in conda list
```
vtk                       8.1.0           py36h9686630_201  
vtk                       8.1.1                     <pip>

```
Then I get the following:
```

$ pip install vtk
Requirement already satisfied: vtk in /home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages (8.1.1)
mkl-random 1.0.1 requires cython, which is not installed.
mkl-fft 1.0.2 requires cython, which is not installed.
You are using pip version 10.0.1, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

```
however, cython is installed
```
$ which cython
/usr/bin/cython
```

larsoner, 2018-09-04:
If you got that message at the pip install step, you must have skipped the conda uninstall (maybe remove) and/or pip uninstall steps because these should ensure that vtk is no longer installed

dnacombo, 2018-09-04:
my bad.
Now reinstalled.
Running my test with F5:

```
An error ocurred while starting the kernel
Qt WebEngine seems to be initialized from a plugin. Please set Qt::AA_ShareOpenGLContexts using QCoreApplication::setAttribute before constructing QGuiApplication.
Fatal Python error: Segmentation fault

Thread 0x00007f2ae774b700 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/zmq/utils/garbage.py", line 47 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f2aee1e9700 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/parentpoller.py", line 39 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f2b067fc700 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 295 in wait
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 551 in wait
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/history.py", line 829 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/history.py", line 58 in needs_sqlite
File "", line 2 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f2b07fff700 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/heartbeat.py", line 61 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f2b0cd84700 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/selectors.py", line 445 in select
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/asyncio/base_events.py", line 1398 in _run_once
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/asyncio/base_events.py", line 422 in run_forever
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/tornado/platform/asyncio.py", line 127 in start
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/iostream.py", line 78 in _thread_main
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 864 in run
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f2b2bf34740 (most recent call first):
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/tvtk/tvtk_base.py", line 568 in _wrap_call
File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/mayavi/tools/figure.py", line 348 in screenshot
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/mne/mne/report.py", line 77 in _fig_to_img
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/mne/mne/report.py", line 922 in _add_figs_to_section
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/mne/mne/report.py", line 967 in add_figs_to_section
File "/home/maximilien.chaumon/owncloud/Lab/Projects/CTPS/code/test.py", line 29 in 
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/spyder_kernels/customize/spydercustomize.py", line 106 in execfile
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/spyder_kernels/customize/spydercustomize.py", line 678 in runfile
File "", line 1 in 
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/interactiveshell.py", line 2963 in run_code
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/interactiveshell.py", line 2909 in run_ast_nodes
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/interactiveshell.py", line 2785 in _run_cell
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/IPython/core/interactiveshell.py", line 2662 in run_cell
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/zmqshell.py", line 537 in run_cell
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/ipkernel.py", line 208 in do_execute
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/kernelbase.py", line 399 in execute_request
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/kernelbase.py", line 233 in dispatch_shell
File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site‑packages/ipykernel/kernelbase.py", line 283 in dispatcher
[... cut ...]
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 77 in _fig_to_img
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 922 in _add_figs_to_section
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mne/mne/report.py", line 967 in add_figs_to_section
  File "test.py", line 29 in <module>
Segmentation fault (core dumped)
```

larsoner, 2018-09-04:
You get these same errors with the minimal example I had above, right? Let's call that 3-line script `minimal_example_fail.py`.

If so, the next steps might be to create a clean env with:
```
$ conda deactivate
$ conda create -n test python=3.6 numpy "pyqt>=5" matplotlib traits pyface traitsui
$ conda activate test
$ pip install vtk
$ pip install mayavi
$ python minimal_example_fail.py
```
If this dies, then we can open a Mayavi issue about it. If it does not fail, then we perhaps you can do some testing to figure out what package difference exists between the `mne` and `test` envs.

One other thing to try is to change `ENABLE_USER_SITE = None` to `False` in `anaconda3/envs/mne/lib/python3.6/site.py` to see if it makes anything better.

dnacombo, 2018-09-05:
minimal_example_fail.py:
```Python
import faulthandler
faulthandler.enable()

#import matplotlib.pyplot as plt; assert plt.get_backend() == 'Qt5Agg'

from mayavi import mlab
mlab.test_plot3d()
mlab.screenshot()

```
```
$ python minimal_example_fail.py 
Fatal Python error: Segmentation fault

Current thread 0x00007f2c3107e740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 568 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "minimal_example_fail.py", line 15 in <module>
Segmentation fault (core dumped)
```

After creating test environment, activating it, successfully installing vtk 8.1.1 and mayavi 4.6.2 (warning about cython not being installed, though it is)...

```
$ python minimal_example_fail.py
Fatal Python error: Segmentation fault

Current thread 0x00007f8575b66740 (most recent call first):
  File "/home/maximilien.chaumon/anaconda3/envs/test/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 568 in _wrap_call
  File "tvtk_classes/open_gl_render_window.py", line 79 in get_pixel_data
  File "/home/maximilien.chaumon/anaconda3/envs/test/lib/python3.6/site-packages/mayavi/tools/figure.py", line 348 in screenshot
  File "minimal_example_fail.py", line 15 in <module>
Segmentation fault (core dumped)
```

agramfort, 2018-09-05:
open an issue on mayavi tracker with the necessary info to replicate.

cc @larsoner and myself on this.

larsoner, 2018-10-08:
Closing this because it's an upstream bug, but please let us know what the solution is if/when it's sorted out

closed 2018-10-08 by larsoner