mne-tools/mne-python #5480: Raw.plot_sensors placing the sensors in the wrong positions

olavblj, 2018-08-29:
#### Describe the bug
When using `plot_sensor` on a small set of sensors, the plot is wrong. It however works when plotting with all the sensors.

#### Steps and/or code to reproduce

```Python
        subject = 1
        runs = [6, 10, 14]  # motor imagery: hands vs feet

        raw_fns = eegbci.load_data(subject, runs)
        raw_files = [read_raw_edf(f, preload=True, stim_channel='auto') for f in
                     raw_fns]
        raw = concatenate_raws(raw_files)

        raw.rename_channels(lambda x: x.strip('.'))
        raw.pick_channels(["C3", "C4", "P3", "P4", "STI 014"]) # If this line is removed, it will plot the sensors positions correctly
        raw.rename_channels({"STI 014": "STIM"})

        raw.set_montage(read_montage("standard_1020"))
        raw.plot_sensors(block=True)
```

#### Expected results
I expected a plot of the correct sensor positions.

#### Actual results
This plot, which is clearly wrong.

[image]

#### Additional information
Paste the output of `mne.sys_info()` here.

```
Platform:      Darwin-17.6.0-x86_64-i386-64bit
Python:        3.6.6 (default, Aug 22 2018, 11:24:38)  [GCC 4.2.1 Compatible Apple LLVM 9.1.0 (clang-902.0.39.2)]
Executable:    [my project path]/venv/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.17.dev0
numpy:         1.14.5 {blas=-msse3, lapack=-msse3}
scipy:         1.1.0
matplotlib:    2.2.3 {backend=module://backend_interagg}

sklearn:       0.19.2
nibabel:       2.3.0
mayavi:        4.6.1 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.4
```

olavblj, 2018-08-29:
That's cool, thanks for the quick reply! Do you have any idea when it will be merged in though?

larsoner, 2018-08-29:
Hopefully in a couple of weeks. Need to finish the work and make sure it does not break anything

olavblj, 2018-09-28:
Hi! Sorry to bother you, but I'm wondering how this patch is coming along. I am considering using MNE in my master thesis, and will be dependent on it working if I choose to do so.

larsoner, 2018-09-28:
It should be done by November. I don't know how soon before then