mne-tools/mne-python #5483: RawArray.plot() plot is blank

mirgee, 2018-08-30:
I am trying to simply plot a `RawArray`. Am I doing something wrong with the following code, or is it a bug? If it's the later, please, let me know which information could help someone fixing it.

```
df = pd.read_table(os.path.join(raw_root, filename), sep='\t', names=names, skiprows=[0])

info = mne.create_info(ch_names=names, sfreq=250, ch_types='eeg')
data = mne.io.RawArray(np.transpose(df.values), info)

data.set_eeg_reference('average', projection=True)
data.apply_proj()
data.plot(block=True, lowpass=40)
```

The output looks like this:
[image]

Seemingly, `data` is correctly initialized instance of `RawArray` - e.g. `data.get_data()` gives:
```
array([[ 1.05263158e-06,  2.40000000e-05,  2.83947368e-04, ...,
        -3.01045300e+00, -2.49277295e+00, -2.75082789e+00],
       [ 6.05263158e-06,  1.19000000e-04,  1.19994737e-03, ...,
        -1.96403200e+00, -3.14232295e+00, -4.17662589e+00],
       [ 4.50526316e-05,  8.56000000e-04,  7.62994737e-03, ...,
        -1.70380300e+00, -4.35836895e+00, -3.86998589e+00],
       ...,
       [-4.94736842e-06, -1.06000000e-04, -9.74052632e-04, ...,
        -2.43709600e+00, -2.45728995e+00, -2.50819789e+00],
       [-5.94736842e-06, -1.17000000e-04, -1.10305263e-03, ...,
         1.31647100e+00,  1.72476205e+00,  1.63376611e+00],
       [-3.94736842e-06, -8.70000000e-05, -8.03052632e-04, ...,
         2.00205200e+00,  2.61161005e+00,  2.76483411e+00]])
```

agramfort, 2018-08-30:
you need to pass data in V if you say it's EEG

so it should be in the range of uV to see something

mmagnuski, 2018-08-31:
The black screen you see is your signal that is incorrectly scaled so it covers every possible pixel :).
You can try `data.plot(block=True, lowpass=40, scalings='auto')`.

mirgee, 2018-08-31:
@mmagnuski Thank you, this helped! Closing.

closed 2018-08-31 by mirgee