mne-tools/mne-python #5484: Add a list of 'related projects' to the documentation

TomDonoghue, 2018-08-30:
In discussion with @agramfort (and related to discussions with @jasmainak ) - we're thinking of adding a page somewhere in the documentation for 'related projects', or that would list projects that can be used as 'plugins' / 'extensions'. 

Examples of things that could be listed like this - 'plug-in' style tools like [autoreject](https://github.com/autoreject/autoreject), and perhaps related tools that can be used in an MNE pipeline, such as [FOOOF](https://github.com/voytekresearch/fooof). 

Questions:
- Does this seem like a good idea?
- Where would it go?

agramfort, 2018-08-30:
I would add a section at the bottom of
http://martinos.org/mne/stable/documentation.html

also add pactools

jasmainak, 2018-08-31:
I agree with the sentiment. I would also add `mne-hcp` and `mne-bids`. They are clear candidates for such a page.

larsoner, 2020-05-27:
Closing, we have this in the `pre_install` page:

https://mne.tools/dev/install/pre_install.html#overview-of-the-mne-tools-suite

@TomDonoghue feel free to open a PR to add FOOOF or other packages

closed 2020-05-27 by larsoner