mne-tools/mne-python #5487: Splitting file after concatenation

Frooob, 2018-09-02:
Hello,

since my professor asked me to first crop all of the recording that is not explicitly recorded during the task. I had to split it into two files to exclude the part of the recording in the break between the end of the first and beginning of the second part of the experiment, then concatenate the two parts to perform ICA Artifact Rejection.
Now i have to split the two parts again for further analysis. Of course I could crop out two parts of the concatenated parts. But this would appear a bit hacky to me and I would be surprised if there was no command for it. I couldn't find one though.
Do you know if there is a way to do so?

And do you maybe know if excluding the part between the two parts of the experiment is even worth its while? Or even excluding the parts before and after starting the task on the computer? 
All concerning precision of ICA and maybe other preprocessing means.

I would be pleased if you could help me with my problem

agramfort, 2018-09-03:
>
> are you talking about Raw, Epochs or Evoked objects?

what type of analysis do you want to do?

Frooob, 2018-09-03:
I want to perform the ICA on raw data(at least I'm supposed to), exclude artifact components, then epoch it and examine averaged Erps(p300). This by a repeated measures Ancova.

agramfort, 2018-09-03:
why do you need to split the data?
note that you can also run ICA on epochs

agramfort, 2018-09-03:
I would edit the events using numpy operations to make sure they don't
overlap and compute epochs

it's simpler.

larsoner, 2018-11-07:
@Frooob let us know if this solution does not work for you and we will reopen

closed 2018-11-07 by larsoner

psychoxiatao, 2018-11-23:
I also have this question. my data is sleep EEG. and I want to exclude some sleep stages data and focus on one stage and do some analysis.

agramfort, 2018-11-26:
@psychoxiatao you may want to look at this WIP PR: https://github.com/mne-tools/mne-python/pull/5718