mne-tools/mne-python #5488: Add reader for Curry Files

agramfort, 2018-09-05:
#### Describe the problem

Python still lacks a reader for Curry file.

Some of the testing files have been converted to Curry format by Michael Wagner
so they can be used for testing. I've put them here:

https://www.dropbox.com/sh/gre3rk71qq1swnh/AABqqh1XNj6WumTKVIUX61TMa?dl=0

Any contribution on this topic is welcome.

DiGyt, 2019-04-26:
@agramfort

I could adapt the MATLAB solution from the EEGLAB loadcurry() extension for this!

agramfort, 2019-04-26:
that would be very useful !

larsoner, 2019-05-24:
Good thing the relicensed under BSD, we couldn't have used their code without it !

DiGyt, 2019-06-13:
@agramfort
After trying countless methods to read the `catch-alp-good-f Curry 7.dat` and `catch-alp-good-f Curry 8.cdt` files from the dropbox files, their data fields just seem completely unrelated to the respective `CTF/catch-alp-good-f.ds` file.
While reading both `catch-alp-good-f Curry` files into read_raw_curry produces the same data as EEGLAB's loadcurry() would, the channels from these files do not correlate at all with any channel from the original CTF file. (I also downloaded the dropbox curry files multiple times to exclude that they might have been corrupted during a download)

Is it possible that the content of the CTF file was updated after the Curry files were created, or that an additional processing step was performed within one of the files?
I already tried stuff like detrending the data, but it still won't yield useful results...

agramfort, 2019-06-13:
can it be the application of the gradient compensation?

>

larsoner, 2019-06-13:
Good idea, look at `raw.compensation_grade` and `raw.apply_gradient_compensation` for the RawCTF instance

DiGyt, 2019-06-13:
But shouldn't they be still more or less linearly related then?
And should this also affect the eeg and misc channel?
If i throw all channels (i think ~150 meg, ~60 eeg and ~30 misc) into a correlation,
and correlate each channel from the ctf data with each channel from any of the read_curry/loadcurry created data fields, i only get one single correlation out of them that is r > 0.999 (which i think might be the stim channel). Even for an r > 0.9 there is only this one channel with a higher correlation.

larsoner, 2019-06-13:
No it should not affect MISC or EEG. *Maybe* EEG somehow ends up with an average ref automatically in some conversion. But the MISC really should not change, modulo some scale factor but the correlation should take care of that.

I think we need some Curry expert to weigh in on this. I don't know anything about Curry, is it a format that is still being used in current machines so we can ask some Curry devs?

DiGyt, 2019-06-13:
I think it should still be used, Curry 8 was released at the end of 2016.
I also don't think that this is a reading problem. I looked up the Curry 7 user guide, which gave some indications on the data parameters in the parameter file:
```
DATA_PARAMETERS START
   DataFormat           = FLOAT
   DataGuid             = {0B86B22B-1A49-411A-9657-4A4BDA3D3ED8}
   DataExtension        = dat
   CurrentSessionLabel  = 
   DataByteOrder        = INTEL
   DataSampOrder        = SAMP
```
The `DataFormat = FLOAT`should be a 4 byte per sample float, `DataByteOrder = INTEL` should mean little-endianness and `DataSampOrder = SAMP` could be either Fortran or C-like order, but probably Fortran.

This is exactly how it is read in `read_curry` and also in `loadcurry()`, and works fine for other files...

However, i think i will for now stop testing on the `catch-alp-good-f` data.
Maybe i can ask people whether they can give me access to curry and create some additional testing files (wouldn't be bad either to create some smaller testing files than `catch-alp-good-f` ).
What do you think?

larsoner, 2019-06-13:
Yes the smallest file possible that allows us to make sure things are working properly is better anyway.

DiGyt, 2019-06-19:
@agramfort @larsoner 

So, after the `CTF/catch-alp-good-f.ds` CTF file resulted in different data fields for Curry and for MNE,
I tried to check it by reading the same CTF into FieldTrip.

While FieldTrip read the other MNE CTF testing files without problems, `catch-alp-good-f.ds` lead to the following errors:
```
Error using ft_datatype_sens (line 396)
inconsistent number of channels in sensor description

Error in ft_datatype_sens (line 158)
    sens = ft_datatype_sens(sens, 'version', '2011v2');

Error in undobalancing (line 58)
sens = ft_datatype_sens(sens);

Error in channelposition (line 34)
  sens = undobalancing(sens);

Error in ft_datatype_sens (line 336)
        [chanpos, chanori, lab] = channelposition(sens);

Error in ft_datatype_sens (line 158)
    sens = ft_datatype_sens(sens, 'version', '2011v2');

Error in ft_read_header (line 2591)
  hdr.grad = ft_datatype_sens(hdr.grad);
```

@larsoner could it be that this file was somehow intended to test some special cases and was therefore modified to some degree?

Or could it be that this file contains some type of data that just can't be read properly by FT and Curry?

larsoner, 2019-06-19:
Files were added here:

https://github.com/mne-tools/mne-testing-data/commit/720e4b52f9fdee5f351cdb7b5b93af0770e69981#diff-33e2bcaa5268561c3ba06eedd2789561

Following a trail of `blame` here

https://github.com/mne-tools/mne-python/blame/master/mne/io/ctf/tests/test_ctf.py#L25-L30

I could find the PR where they were added:

https://github.com/mne-tools/mne-python/pull/2628

So I added these files. I checked my email for this name and found that they came in a `.zip` file from @ftadel on Nov 23 2015. I don't think we modified them other than adding a `.eeg` file later. Does FieltTrip work if you remove that file? If not, Brainstorm should work. It would be good to check against it.

DiGyt, 2019-06-20:
But there is no `.eeg` file in the `catch-alp-good-f.ds` folder. I'm not a CTF expert,  but could the problem be that there is no such file in the folder?

By the way, Brainstorm reads the data in the same way as MNE does.

Actually, if you look at the file on the left created from mne_read_curry (which is the same is Curry or CTF read in CUrry) to the file on the right, created from read_raw_ctf, there are some strange things in there:

[image]

First thing, is that for the curry file, the time should be constant, but looks more like an eeg channel. Then, I'm not sure, why for the first 30 of the EEG chans, the signal should be so much weaker than for the last 30 EEG chans. Then, for the read_raw_ctf file, the BG/BP/BR channels look quite useful, while for the curry file, BR1 (the one marked as bad), is extremely out of tune with all other chans there.

As said, the read_raw_curry looks very much like the Curry/CTF files Michael Wagner showed us (except that Curry plots the data in "med mode", i.e. with negative values plotted upwards) and has the same initial values.
So i'd say, the most plausible thing would be that there's something in `catch-alp-good-f.ds`, that leads FieldTrip/Curry to throw errors on these files or read them wrong. Could that be?

larsoner, 2019-06-20:
It's possible. I'd try to find a dataset where FT and MNE give the same result and work from there instead of continuing to struggle with this dataset.

massich, 2019-07-22:
closed via #6272

closed 2019-07-22 by massich