mne-tools/mne-python #5492: ENH: Refactor plot_topomap for arrowmaps

SherazKhan, 2018-09-06:
I am wondering, how I can refactor plot_topomap for arrowmaps ?

[image]

agramfort, 2018-09-06:
give it a try. You can have arrows with
https://matplotlib.org/api/_as_gen/matplotlib.axes.Axes.quiver.html

SherazKhan, 2018-09-06:
It can work on any system, we can take virtual gradients of magnetometer or axial gradiometer data.

Currently, I am transforming ctf data to a virtual 283 gradiometer system.

SherazKhan, 2018-09-25:
@larsoner @agramfort 

Please review the code below and let me know, how can I integrate this code in the mne-python API

[image]

```
import mne
from mne.viz import plot_topomap
from mne.viz.topomap import _check_outlines
from mne.datasets import sample, spm_face
from mne.io.pick import pick_types, pick_info
from mne.forward._field_interpolation import _map_meg_channels
from mne.forward._make_forward import _read_coil_defs
from mne.channels.layout import _find_topomap_coords
import matplotlib.pylab as plt
from matplotlib.tri import Triangulation, CubicTriInterpolator
import os.path as op
import numpy as np
plt.ion()

def trigradient(x, y, z):
    tri = Triangulation(x.ravel(), y.ravel())
    tci = CubicTriInterpolator(tri, z.ravel())
    dx, dy = tci.gradient(tri.x, tri.y)
    return dx, dy, tri

sample_path = sample.data_path()
spm_face_path = spm_face.data_path()
templates = _read_coil_defs(verbose=False)

evoked_neuromag= mne.read_evokeds(op.join(sample_path,'MEG','sample' ,
                                          'sample_audvis-no-filter-ave.fif'))

raw_ctf = mne.io.read_raw_ctf(op.join(spm_face_path,'MEG','spm' ,
                                      'SPM_CTF_MEG_example_faces1_3D.ds'))
raw_ctf.apply_gradient_compensation(0)

pick_from = pick_types(evoked_neuromag[0].info, meg='grad', eeg=False,
                       ref_meg=False, exclude=[])
pick_to = pick_types(raw_ctf.info, meg=True, eeg=False, stim=False,misc=False,
                     ref_meg=False, exclude=[])

info_from = pick_info(evoked_neuromag[0].info, pick_from, copy=True)
info_to = pick_info(raw_ctf.info, pick_to, copy=True)

mapping = _map_meg_channels(info_from, info_to, mode='accurate')
data = np.dot(mapping, evoked_neuromag[0].data[pick_from])
pos = _find_topomap_coords(raw_ctf.info, pick_to)
fig, ax = plt.subplots()
plot_topomap(data[:, 174], pos, axes=ax)

x, y = tuple(pos.T)
dx, dy, tri = trigradient(x, y, data[:,174])

dxx = dy.data
dyy = -dx.data

pos, outlines = _check_outlines(pos, 'head', None)
#ax.triplot(tri, 'ko', lw=0)
x, y = tuple(pos.T)
ax.quiver(x, y, dxx, dyy, scale=0.0000000003, color='blue')
```

dengemann, 2018-09-25:
Looks pretty neat! Perhaps play a bit with the arrow color. I'd try black
or gray.

SherazKhan, 2018-09-25:
[image]

agramfort, 2018-09-25:
here the quiver are derived from the data passed to plot_topomap. If you
want to support this feature I would add a bool parameter to plot_topomap
to show or not the arrows. If you want to pass any field vector you need to
pass a 2D array as a new arrows parameter.

SherazKhan, 2018-09-25:
> here the quiver are derived from the data passed to plot_topomap. If you want to support this feature I would add a bool parameter to plot_topomap to show or not the arrows. If you want to pass any field vector you need to pass a 2D array as a new arrows parameter.

I think refactoring plot_topomap is cumbersome, may be including plot_arrowmap and trigradient functions in the topomap module is the best best.

If you are ok, I can sent the PR.

```
import mne
from mne.viz import plot_topomap
from mne.viz.topomap import _check_outlines, _prepare_topo_plot
from mne.datasets import sample
from mne.io.pick import pick_types, pick_info, _picks_by_type
from mne.forward._field_interpolation import _map_meg_channels
import matplotlib.pylab as plt
from matplotlib.tri import Triangulation, CubicTriInterpolator
import os.path as op
import numpy as np

plt.ion()

def trigradient(x, y, z):
    """Prepare gradient of z on the mesh define by x and y"""
    tri = Triangulation(x.ravel(), y.ravel())
    tci = CubicTriInterpolator(tri, z.ravel())
    dx, dy = tci.gradient(tri.x, tri.y)
    return dx, dy

def plot_arrowmap(data, info_from, info_to=None):

    """Plotting arrow map
    ----------
    data : array, shape (n_chan,)
        The data values to plot.
    info_from : instance of Info
        The measurement data to interpolate from.
    info_to : instance of Info
        The measurement info to interpolate to.
    Returns
    -------
    im : matplotlib.image.AxesImage
        The interpolated data.
    cn : matplotlib.contour.ContourSet
        The fieldlines..
    """

    ch_type = _picks_by_type(info_from)
    if len(ch_type) > 1:
        raise ValueError("Multiple Channel type not supported")
    else:
        ch_type = ch_type[0][0]

    if ch_type not in ('mag', 'grad'):
            raise ValueError("Channel type not supported. Supported channel "
                         "types include 'mag' and 'grad'.")

    if info_to is None and ch_type is 'mag':
        info_to = info_from
    else:
        ch_type = _picks_by_type(info_to)
        if len(ch_type) > 1:
            raise ValueError("Multiple Channel type not supported")
        else:
            ch_type = ch_type[0][0]

        if ch_type not in ('mag'):
            raise ValueError("only mag Channel type is supported")

    if info_to is not info_from:
        mapping = _map_meg_channels(info_from, info_to, mode='accurate')
        data = np.dot(mapping, data)

    pos = _prepare_topo_plot(info_to, ch_type='mag', layout=None)[1]
    pos = _check_outlines(pos, 'head', None)[0]
    fig, ax = plt.subplots()
    im, cn = plot_topomap(data, pos, axes=ax)
    x, y = tuple(pos.T)
    dx, dy = trigradient(x, y, data)
    dxx = dy.data
    dyy = -dx.data
    ax.quiver(x, y, dxx, dyy, scale=0.0000000003, color='k', lw=2)
    return fig, ax, im, cn

sample_path = sample.data_path()
evoked_neuromag = mne.read_evokeds(op.join(sample_path,'MEG','sample' ,
                                          'sample_audvis-no-filter-ave.fif'))

pick_from = pick_types(evoked_neuromag[0].info, meg='grad', eeg=False,
                       ref_meg=False)
pick_to = pick_types(evoked_neuromag[0].info, meg='mag', eeg=False, stim=False,misc=False,
                     ref_meg=False, exclude=[])
info_from = pick_info(evoked_neuromag[0].info, pick_from, copy=True)
info_to = pick_info(evoked_neuromag[0].info, pick_to, copy=True)

data = evoked_neuromag[0].data[pick_from,  174]
fig, ax, im, cn = plot_arrowmap(data, info_from, info_to)
```

[image]

agramfort, 2018-09-26:
fine with me. However I think you only need one info

can you open a PR?

SherazKhan, 2018-09-26:
The reason for second info, is to able to interpolate this map for the neuromag gradiometers, as they take gradients in circular coordinates i.e. latitude and longitude. Therefore need to be projected for visualization on the 2D manifold like topomap.

I will push the PR

SherazKhan, 2018-09-26:
PR done, one of many that will follow this year :-)

larsoner, 2018-11-07:
We have this now thanks to @SherazKhan :)

closed 2018-11-07 by larsoner