mne-tools/mne-python #5494: RuntimeError: EEG 61 or EEG 62 channel not found !!

LeilaNS, 2018-09-07:
Hi,
I am trying to do autoreject ICA, the raw file dosen't hav EOG 61, 62. Although my raw files has EEG061 and EEG062, I got the this error " RuntimeError: EEG 61 or EEG 62 channel not found !!"

- [ ] in _get_eog_channel_index
- [ ]     raise RuntimeError('EEG 61 or EEG 62 channel not found !!')
- [ ] 
- [ ] RuntimeError: EEG 61 or EEG 62 channel not found !!
- [ ] 
- [ ] 

- [ ] In [6]: raw.ch_names[369], raw.ch_names[370]
- [ ] Out[6]: (u'EEG062', u'EEG063')
Can this issue be solved?
Thanks,

LeilaNS, 2018-09-07:
> what function are you using? It's hard to guess what's wrong without seeing
> the full code snippet.

picks_eog = np.concatenate([picks, mne.pick_types(raw.info, meg=False,

eeg=False, ecg=False,

eog=True)])

# Create eog-based epochs

eog_epochs = create_eog_epochs(raw, reject=reject, picks=picks_eog,

baseline = (None, 0), tmin=-0.1, tmax=0.2)

eog_average = eog_epochs.average()

# Find eog artifacts based on correlation between ICA and EOG061/EOG062

eog_inds, scores = ica.find_bads_eog(eog_epochs)

The error comes from the last line.

Thanks

agramfort, 2018-09-07:
@LeilaNS can you check that you do have channels marked as EOG channels?

also you can specify `ch_name` parameter in the `find_bads_eog` method.

So for your data something like:

`eog_inds, scores = ica.find_bads_eog(eog_epochs, ch_name='EOG061')`

should work.

LeilaNS, 2018-09-07:
I don't have any channels marked as EOG.

In [45]: eog_inds, scores = ica.find_bads_eog(eog_epochs,ch_name='EEG061')

in _get_eog_channel_index

raise ValueError('%s not in channel list' % ch_name)

ValueError: ['EEG061'] not in channel list

In [46]: raw.ch_names[368]

Out[46]: u'EEG061'

agramfort, 2018-09-07:
do you have a channel called EOG061 ?

find_bads_eog detects bad components by seeing how much they are time
locked with the EOG channel. You must tell him what channel to use for
EOG.

LeilaNS, 2018-09-07:
No I don't

agramfort, 2018-09-07:
you can try by passing a frontal channel. Maybe it's good enough

LeilaNS, 2018-09-07:
I recorded the MEG data by EOG electrodes but they marked as 'EEG061' and
'EEG062' so I would like to use EEG061as EOG061.
How can I change the ch_names?
Thanks

agramfort, 2018-09-07:
https://martinos.org/mne/dev/generated/mne.io.Raw.html#mne.io.Raw.rename_channels

LeilaNS, 2018-09-07:
> io.Raw.rename_channels

I do:
raw.rename_channels(mapping={'EEG061': u'EOG061'})
raw.rename_channels(mapping={'EEG062': u'EOG062'})

picks_eog = np.concatenate([picks, mne.pick_types(raw.info, meg=False,
eeg=False, ecg=False, eog=True)])

eog_epochs = create_eog_epochs(raw, reject=reject, picks=picks_eog,
baseline = (None, 0), tmin=-0.1, tmax=0.2)

And still the same error :
No EOG channels found
Trying with EEG 061 and EEG 062
Traceback (most recent call last):

in _get_eog_channel_index

raise RuntimeError('EEG 61 or EEG 62 channel not found !!')

RuntimeError: EEG 61 or EEG 62 channel not found !!

In [18]: raw.ch_names[368], raw.ch_names[369]

Out[18]: (u'EOG061', u'EOG062')

agramfort, 2018-09-07:
changing the channel names does not change the channel types. You need to
use:

https://martinos.org/mne/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types

LeilaNS, 2018-09-12:
> io.Raw.set_channel_types

Problem fixed, thanks.

closed 2018-09-12 by agramfort