mne-tools/mne-python #5499: [BUG] meas_date: list, int, tuple?

sappelhoff, 2018-09-09:
This is related to #5432 but reports a concrete bug that needs to be solved regardless on how the overarching discussion in #5432 goes.

meas_date should be a list of two integers: 

```
meas_date : list of int

    The first element of this list is a UNIX timestamp (seconds since 1970-01-01 00:00:00)
denoting the date and time at which the measurement was taken.
The second element is the additional number of microseconds.

```

I found two issues / bugs:

1. `DATE_NONE` is a tuple, not a list:

https://github.com/mne-tools/mne-python/blob/613df87f097323175784f3a7fdc32bbad0e604ac/mne/io/write.py#L25

2. For some data readers, `meas_date` is formatted as an integer and not a list of ints.

```python

import os.path as op
import mne

base_path = op.join(op.dirname(mne.__file__), 'io')

data_path = op.join(base_path, 'kit', 'tests', 'data')
raw_fname = op.join(data_path, 'test.sqd')

raw = mne.io.read_raw_kit(raw_fname)

# BUG --> this should return a list, but returns int
raw.info['meas_date']
```

The output: 

```
Extracting SQD Parameters from /home/stefanappelhoff/Desktop/mne-python/mne/io/kit/tests/data/test.sqd...
Creating Raw.info structure...
Setting channel info structure...
Creating Info structure...
Current compensation grade : 0
Ready.

1338487633
```

I found this while developing MNE-BIDS in https://github.com/mne-tools/mne-bids/pull/78, where we test a multitude of formats (EEG and MEG) in conjunction.

sappelhoff, 2018-09-09:
Why would you choose tuple @larsoner? Most of the codebase and documentation seems to handle meas_date as a list.

As a first easy step I would propose changing

https://github.com/mne-tools/mne-python/blob/613df87f097323175784f3a7fdc32bbad0e604ac/mne/io/write.py#L25

to a list instead of a tuple.

larsoner, 2018-09-09:
Because tuple is immutable and this is not something meant to be manually changed by a user

larsoner, 2018-09-09:
See #5500

PR #5500 (2018-09-10): [MRG+2] Unify meas_date (to use a tuple)
Closes #5499.
commit d1e2cc88 (2018-09-10): [MRG+2] Unify meas_date (to use a tuple) (#5500)

Closes #5499.
closed 2018-09-10 by massich