mne-tools/mne-python #5501: Feature request: incremental updates to an `mne.Report`

wmvanvliet, 2018-09-10:
This is something that came up when writing the huge analysis pipeline for the [conpy](https://github.com/wmvanvliet/conpy) project:

**I think the `Report` object of MNE-Python could be made even more useful if we could write incremental updates to it.**

As the analysis progresses, more and more figures are incrementally added to the report. Every time the report is saved, the `.html` file is updated. This would make it super easy to add visual checks to each step of the analysis. Easier than the current situation where we have to make one giant `make_report.py` file at the end of the pipeline that reads in all the intermediate results and plots them. 

Say I have three analysis scripts:

    01_filter_raw.py
    02_make_epochs.py
    03_grand_average.py

It would be cool if each of these scripts could add/update figures in a shared `Report` object, for example like this:

```python
# Somewhere in 01_filter_raw.py
report = mne.read_report('subject1_report.pickle')  # Loads existing or creates new
report.add_figs_to_section([raw.plot_psd()], ['PSD plot'], 'preprocessing')
report.pickle()  # since report.save() is already taken

# Somewhere in 02_make_epochs.py
report = mne.read_report('subject1_report.pickle')
report.add_figs_to_section([epochs.average.plot()], ['Evokeds'], 'preprocessing')
report.pickle()

# Somewhere in 03_grand_average.py
report = mne.read_report('subject1_report.pickle')
report.add_figs_to_section([grand_average.plot()], ['Grand average'], 'preprocessing')
report.pickle()
```

Even implement `__enter__()` and `__exit__()` blocks so we can do this:

```python
with mne.read_report('subjec1_report.pickle') as report:
     report.add_figs_to_section(...)
# report is automatically saved when exiting the block
```

For this to be used as intended, there needs to be a way to overwrite existing figures.

agramfort, 2018-09-10:
can we already pickle a report object? if so it should quite easy to
implement

wmvanvliet, 2018-09-10:
yes we can! I can quickly create a PR with the bulk of the desired functionality if there seems to be interest in this feature.

agramfort, 2018-09-10:
I don't think it's much code and I can understand the practical value. So
let's try I would say

larsoner, 2018-09-10:
From what I understand pickling is a bad way to get long-term persistence. I'd rather use `hdf5`, in theory it shouldn't add too many lines beyond what the `pickle` does.

larsoner, 2018-09-10:
Thinking even smaller scale, `pickle` seems like an unreliable way to transfer data across Python instances. Say for some reason (mayavi!) a user ran one script in Python 3 env to do preproc, then later a different one in Python 2 to viz source estimates, a `pickle` pipeline very likely will fail. These sorts of corner cases should be pretty easily avoided via `h5io` storage. It's more standard for us anyway so it's also more consistent.

The real question is what to call this. Maybe `Report.save(fname, output='html' | 'instance')`?

wmvanvliet, 2018-09-10:
`Report.to_hdf5`?

larsoner, 2018-09-10:
`.to_*` makes it sound like it returns an object. We usually use `write_` or `save` for things that write to disk, and for methods it's always `save` of some sort -- so perhaps `save_hdf5`?

jasmainak, 2018-09-10:
Since the saved report is an html, wouldn't it be possible to just read it and grep for the `<div>` elements, populate the `self.html` list that's internal to the `Report` object? It's probably more work than the `save_hdf5` option though ...

wmvanvliet, 2018-09-11:
> maybe use save and decide what to use based on file extension?

that's a good idea!

PR #5505 (2018-09-14): [MRG] Add open/save to HDF5 ability to mne.Report
This allows you to save `mne.Report` objects in such a way that they can be read back later. This in turn allows you to incrementally add figures to a `Report` object across scripts.

The `mne.open_report` function either reads a `Report` if the file exists, or creates a new one if it doesn't. This mimics the behavior of the build-in Python `open` function. You can use it
like this:

    report = mne.open_report('report.h5')
    report.add_figures_to_section(...)
    report.save('report.html')  # by default, it saves the report back to 'report. [... cut]
closed 2018-09-14 by larsoner