mne-tools/mne-python #5508: mne coreg fails

subhuti2, 2018-09-11:
[test.elp.txt](https://github.com/mne-tools/mne-python/files/2371726/test.elp.txt)

Dear mne_analysers,

first thank you for the great software. I am using the python version (0.16.2@mac os, with python3, anaconda3) of mne, everything was installed according to MNE instruction. I have some problems with the coregistration (command mne coreg). I started with a template mri and Neuroscan EEG 3dd file for locations of the EEG electrodes and fiducials. I did the following:
1 by some conversions, I generate an ascii file with 3 columns of data, indicating x, y, z coordinates of the fiducials and electrodes, which was saved with elp extension.The file was renamed and attached. 
2 This file was readin and saved with the following two lines:
cht = mne.channels.read_dig_montage (elp='/thePath/theFile.elp', point_names=('nasion', 'lpa', 'rpa' ,'chan1', 'chan2', 'chanN');  
cht.save('/thePath/theFile.fif');
3 I start mne coreg, I load the template mri image marked with fiducials, then load "theFile.fif".

I got the error in the last step:
   File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482, in dispatch
    handler( *args )
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 965, in wrapper0
    return function(arg)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 3371, in notify
    self.trait_property_changed( name, old )
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 895, in decorator
    self.__dict__[ name ] = result = function( self )
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_file_traits.py", line 309, in _get__info
    for idx, pos in enumerate(info.hsp):
TypeError: 'NoneType' object is not iterable
Coregistration: Excluding 0 head shape points with distance >= inf m.
^CTraceback (most recent call last):
  File "/UserPath/anaconda3/envs/mne/bin/mne", line 39, in <module>
    sys.exit(subprocess.call([sys.executable, cmd_path] + sys.argv[2:]))
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/subprocess.py", line 269, in call
    return p.wait(timeout=timeout)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/subprocess.py", line 1457, in wait
    (pid, sts) = self._try_wait(0)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/subprocess.py", line 1404, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)

Do you think it is a data problem (the electrodes locations were not correctly read and written) or some installation problem?

thanks in advance,

Peng

Detailed instructions on how to file a bug can be found in our [FAQ](https://martinos.org/mne/stable/faq.html#i-think-i-found-a-bug-what-do-i-do).

If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the bug
Please provide a clear and concise description of the bug.

#### Steps and/or code to reproduce
Please provide a code snippet or [minimal working example (MWE)](https://en.wikipedia.org/wiki/Minimal_Working_Example)
to replicate your problem.

This MWE should be self-contained, which means that other MNE-Python contributors
should be able to copy and paste the provided snippet and replicate the bug.
If possible, use MNE-Python testing examples to reproduce the error. Otherwise,
provide a small and anonymized portion of your data required to reproduce the bug.

If the code is too long, feel free to put it in a [public gist](https://gist.github.com) and link
it in the issue.

Example:

```Python
import mne

fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
raw = mne.concatenate_raws([raw])
raw.save('test_raw.fif', overwrite=True)
raw_read = mne.io.read_raw_fif('test_raw.fif')  # this breaks
```

#### Expected results
Provide a clear and concise description of what you expected to happen.

#### Actual results
Please paste or specifically describe the actual output or traceback.

#### Additional information
Paste the output of `mne.sys_info()` here.

agramfort, 2018-09-11:
@subhuti2 to know if it's pb of data or system install can you confirm that it fails also with MNE sample dataset on your machine?

subhuti2, 2018-09-12:
Thanks @agramfort , if I load the sample meg dataset (sample_audvis_raw.fif), it seems fine. Then it seems more likely to be a data issue. I think as the 3-column txt file is quite clear, probably the read and/or write function was not as expected. After read in (montage name 'cht'), if I type the name in python, I got the following.
<DigMontage | 0 extras (headshape), 129 HPIs, 3 fiducials, 0 channels>
Then if I try cht.plot(scale_factor=1, show_names=1, kind='3d', show=1)
I got lots of errors, as the following:
Traceback (most recent call last):
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/backends/backend_macosx.py", line 83, in _draw
    self.figure.draw(renderer)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/figure.py", line 1475, in draw
    renderer, self, artists, self.suppressComposite)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/image.py", line 141, in _draw_list_compositing_images
    a.draw(renderer)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mpl_toolkits/mplot3d/axes3d.py", line 281, in draw
    reverse=True)):
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mpl_toolkits/mplot3d/axes3d.py", line 280, in <lambda>
    key=lambda col: col.do_3d_projection(renderer),
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mpl_toolkits/mplot3d/art3d.py", line 462, in do_3d_projection
    ecs = (zalpha(self._edgecolor3d, vzs) if self._depthshade else
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mpl_toolkits/mplot3d/art3d.py", line 769, in zalpha
    colors = get_colors(colors, len(zs))
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/mpl_toolkits/mplot3d/art3d.py", line 760, in get_colors
    (num, 4))
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/cbook/_backports.py", line 147, in broadcast_to
    return _broadcast_to(array, shape, subok=subok, readonly=True)
  File "/UserPath/anaconda3/envs/mne/lib/python3.6/site-packages/matplotlib/cbook/_backports.py", line 101, in _broadcast_to
    op_flags=[op_flag], itershape=shape, order='C').itviews[0]
ValueError: operands could not be broadcast together with remapped shapes [original->remapped]: (129,4) and requested shape (0,4)
<Figure size 1280x1280 with 1 Axes>

I don't know how to extract the 3d sensor locations info from the raw data. As mentioned above, it worked with the mne coreg, thus it must be there. But can not see it either with or without the preload:
<Raw  |  sample_audvis_raw.fif, n_channels x n_times : 376 x 166800 (277.7 sec), ~482.1 MB, data loaded>

agramfort, 2018-09-12:
ok then we have a path to fix the issue. you should share a script that
replicates the issue
as we should see the pb too

subhuti2, 2018-09-13:
1 readin the file test.elp (attached as test.elp.txt), not errors
`cht = mne.channels.read_dig_montage(elp='/UserPath/test.elp', point_names=('nasion', 'lpa', 'rpa', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26', '27', '28', '29', '30', '31', '32', '33', '34', '35', '36', '37', '38', '39', '40', '41', '42', '43', '44', '45', '46', '47', '48', '49', '50', '51', '52', '53', '54', '55', '56', '57', '58', '59', '60', '61', '62', '63', '64', '65', '66', '67', '68', '69', '70', '71', '72', '73', '74', '75', '76', '77', '78', '79', '80', '81', '82', '83', '84', '85', '86', '87', '88', '89', '90', '91', '92', '93', '94', '95', '96', '97', '98', '99', '100', '101', '102', '103', '104', '105', '106', '107', '108', '109', '110', '111', '112', '113', '114', '115', '116', '117', '118', '119', '120', '121', '122', '123', '124', '125', '126', '127', '128'), unit='mm');
`
2. save the file as fif format (no errors in saving, but errors while loaded by mne coreg)
`cht.save('/UserPath/theFile.fif')
`
3. plot the electrodes (generate errors)
`cht.plot(scale_factor=1, show_names=1, kind='3d', show=1)
`

agramfort, 2018-09-13:
please share a dropbox with a full script to replicate the pb.

subhuti2, 2018-09-13:
Dear Alex,
[test.elp.txt](https://github.com/mne-tools/mne-python/files/2378483/test.elp.txt)
sorry that I am in a conference, where dropbox does not work and internet is slow. However the code in the last post can replicate the problem already (You can use any template head mri in mne coreg, e.g. fsaverage). I try again in this post to upload test.elp (as test.elp.txt), which failed in the last post.
Thanks for looking at it.

agramfort, 2018-09-13:
@subhuti2 the cht does not contain any channel:

```
In [9]: cht
Out[9]: <DigMontage | 0 extras (headshape), 128 HPIs, 3 fiducials, 0 channels>
```

but I am a bit clueless on what to advice. @teonbrooks my expert of Montage
and DigMontage what do you think?

subhuti2, 2018-09-14:
@agramfort @teonbrooks  Thanks.
In the file there were 128 locations (electrodes) besides 3 fiducials, it was termed as HPIs as in the data structure (from alex's comments I guess it should be channels).  I changed elp to hpi or hsp, and even nothing (only file name), the locations were always treated as HPIs...

teonbrooks, 2018-09-20:
ah, i see that you are using the elp file format. that format is properly supported with the `DigMontage` class. you should be using the `read_montage` function that works with `Montage` class.

@subhuti2 try using `mne.channels.read_montage(test.elp)`

when I created the `DigMontage`, I used `elp` arg name because that was how we in my lab referred the the electrode location points in native headspace as opposed to `hpi` which is the head point indicator in MEG system space.

I am actually mulling over the way we use digitized points and planning to implement something that would be easier to use, understand and unified.

subhuti2, 2018-09-25:
Thank you @teonbrooks, however it did not work with mne.channels.read_montage either. I quote the error message as within [] .
[TypeError: loadtxt() got an unexpected keyword argument 'skip_header']

In fact I have the information as Nx3 numbers, N is the sum of all electrodes and fiducials. I could also include labels as another column in the file, when needed. But I don't know exactly what format was needed ...

teonbrooks, 2018-09-25:
@subhuti2, for the `read_montage` to work, could you remove the .txt extension that is attached to your .elp file and try that?

subhuti2, 2018-09-26:
@teonbrooks, I did that, but did not help. Is it possible for you to provide an example elp file? Ideally the locations of electrodes could be in xyz format (Cartesian coordinates).
Thanks in advance.

larsoner, 2020-05-04:
I'll close since this should now work with newest MNE. This error:

> TypeError: loadtxt() got an unexpected keyword argument 'skip_header'

should be fixed nowadays, and the new montage functions should help with this use case (hopefully!)

closed 2020-05-04 by larsoner