mne-tools/mne-python #5510: Difficulties localizing resting power

bloyl, 2018-09-12:
I was trying to replicate the brainstrom tutorials (https://neuroimage.usc.edu/brainstorm/Tutorials/RestingOmega) which localizes a alpha, beta power etc and was unable to do it using `compute_source_psd`. I'm not sure if i did something wrong or if there is a bug in `compute_source_psd` somewhere?

i put together an example using the bst_resting dataset. Because the bst_resting dataset supplied by in mne_data is incomplete, the example is a bit involved so it is in a dropbox folder described below.

I'd be happy to turn this into a proper example and fix the bst_resting dataset (#4184) once its figured out.

Thanks...

the link to the dropbox is
https://www.dropbox.com/sh/1ubtsdj2b0pyxqy/AAD34ohc_KrdpEKiffL7EOT9a?dl=0

it contains:
bst_resting_example_2.py - example script

result images
1. beta_bst_resting_brainstrom.png      -- expected beta band results from brainstorm analysis
2. bst_resting_sensor_power.png          -- sensor psd results look fine.
3. bst_resting_beta_mne_dorsal.png     -- beta power localizes to the bottom of the brain
4. bst_resting_beta_mne_ventral.png
 
data files needed for analysis (can also be recreate with example script)
1. subj002_spontaneous_20111102_01_AUX-raw.fif
2. subj002_spontaneous_20111102_01_AUX-empt-cov.fif
3. bst_resting-bem.sol
4. bst_resting-oct6-src.fif
5. bst_resting-trans.fif

larsoner, 2018-09-12:
@dengemann IIRC you worked with this data a bit, can you comment?

larsoner, 2018-09-12:
Also maybe @SherazKhan ?

dengemann, 2018-09-12:
@bloyl I worked recently on an example that shows resting state source results for the brainstorm data. I did not open the pull request because the results did not make sense to me. It seemed that the data has a very weakly pronounced alpha and beta peak and localizing the data sort of wiped out the spatial effects. I could open the PR and we take it from there?

bloyl, 2018-09-12:
That sounds like a good idea.

I agree that the alpha and beta peaks aren't prominent, but the sensor topos look reasonable and brainstorm has no issues giving meaningful/sensible results.

dengemann, 2018-09-12:
@bloyl see #5514

PR #5692 (2018-11-08): [MRG+1]: Add VectorView and OPM rest example
Follow-up of #5514.
Closes #5510.

Implements the Brainstorm pipeline for resting state data. Looks good and seems robust to me on VectorView and OPM data.

Marked WIP pending verification from @bloyl that it looks similar in Brainstorm and/or that @dengemann can replicate on mne-hcp data. @ktavabi has reported that it works for some of his VectorView data.
closed 2018-11-08 by larsoner