mne-tools/mne-python #5512: Make raw.annotations a property with setter decorator

cbrnr, 2018-09-12:
I don't know when this changed, but now there is a `set_annotations` method that needs to be called to set annotations. A more Pythonic way would be to make `raw.annotations` a property with associated setter using the `@property` and `@annotations.setter` decorators.

If people agree, I can create a PR with the changes.

cbrnr, 2018-09-12:
I like properties exactly because of this feature. Especially if it is just a simple check if all values are within bounds, what would be so bad about getting an exception when setting `raw.annotations = my_annotations`?

agramfort, 2018-09-12:
yes I asked for the set_annotations to be explicit that calling the method does something to the assign annotations. It's not just checking that it's valid. It's less magic as it is now.

closed 2018-09-12 by agramfort