mne-tools/mne-python #5519: [BUG] split_naming logic for files with one part

teonbrooks, 2018-09-13:
related to PR https://github.com/mne-tools/mne-python/pull/5470, I just realized now that this split naming solution works only if the file will be split into multiple parts.

I noticed that the way that we decide if the raw file is to be split happens at https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L2326. this is fine for neuromag-style naming convention where there is no suffix appended to the first file, but for BIDS, `part-` is appended only if there is at least a second part.

@larsoner, what do you suggest in terms of including this logic. we would essentially need to add 
```py
if pos >= split_size - this_buff_size_bytes - next_file_buffer and \
                first + buffer_size < stop:
```
before https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L2218 but those variables aren't defined until later

larsoner, 2018-09-13:
I don't see a clean solution because we can't easily know the sizes ahead of time (need to write meas info, full buffers, blank buffers, etc.).

How about for BIDS, whenever writing you:

1. ensure that the unsplit-name and split-name are both available before writing, if they aren't, raise an error (unless `overwrite=True`)
2. if the file needs to be split, `os.rename` the first file to have the `_part` stuff, possible after an `os.remove` if `overwrite=True`.

teonbrooks, 2018-09-27:
oh ok, so this will work fine if the user
1. use the `mne-bids` for saving bids files. the naming split won't auto generate the correct bids filename, just the extension name attached.
2. we can introspect the files to know if the split naming is necessary so it would skirt the issue above.

I will patch this in mne-bids

teonbrooks, 2018-10-10:
closing. bug fix in https://github.com/mne-tools/mne-python/pull/5570 and implemented in https://github.com/mne-tools/mne-bids/pull/102

closed 2018-10-10 by teonbrooks