mne-tools/mne-python #5521: Issue with traits handler in mne.gui.coregistration

gladiach, 2018-09-13:
Hi,

I'm using mne 0.16.2 with python 2.7.11 at the Martinos center.
When running mne.gui.coregistration I get an error related to 'traits':

In: mne.gui.coregistration(inst=fname, subject=subject, subjects_dir=subjects_dir)

Out:
Exception occurred in traits notification handler for object: <traits.traits_listener.ListenerParser object at 0x7fb248e50f50>, trait: text, old value: , new value: mri_head_t,parameters[],coord_frame
Traceback (most recent call last):
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1302, in _text_changed
    self.listener = self.parse()
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1172, in parse
    return self.parse_group( EOS )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1183, in parse_group
    items.append( self.parse_item( terminator ) )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1262, in parse_item
    self.error( "Expected '[]' at the end of an item" )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1291, in error
    ( msg, self.index, self.text ) )
TraitError: Expected '[]' at the end of an item at column 24 of 'mri_head_t,parameters[],coord_frame'
Exception occurred in traits notification handler for object: <traits.traits_listener.ListenerParser object at 0x7fb248e50f50>, trait: text, old value: , new value: mri_head_t,parameters[],coord_frame
Traceback (most recent call last):
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1302, in _text_changed
    self.listener = self.parse()
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1172, in parse
    return self.parse_group( EOS )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1183, in parse_group
    items.append( self.parse_item( terminator ) )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1262, in parse_item
    self.error( "Expected '[]' at the end of an item" )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 1291, in error
    ( msg, self.index, self.text ) )
TraitError: Expected '[]' at the end of an item at column 24 of 'mri_head_t,parameters[],coord_frame'
Traceback (most recent call last):
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 3066, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-144-4799503a5d5b>", line 8, in <module>
    mne.gui.coregistration(inst=rawfname, subject=subject, subjects_dir=subjects_dir)
  File "<string>", line 2, in coregistration
  File "/homes/8/ghotan/.local/lib/python2.7/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/homes/8/ghotan/.local/lib/python2.7/site-packages/mne/gui/__init__.py", line 170, in coregistration
    scale=scale)
  File "/homes/8/ghotan/.local/lib/python2.7/site-packages/mne/gui/_coreg_gui.py", line 1794, in __init__
    head_high_res=head_high_res)
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 3510, in _init_trait_listeners
    getattr( self, '_init_trait_%s_listener' % data[0] )( name, *data )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 3565, in _init_trait_delegate_listener
    self.on_trait_change( notify, name_pattern, target=self )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 2814, in on_trait_change
    listener.register( self )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 433, in register
    value = getattr( self, type )( new, name, False )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/traits_listener.py", line 701, in _register_simple
    return next.register( getattr( object, name ) )
  File "/homes/8/ghotan/.local/lib/python2.7/site-packages/mne/gui/_coreg_gui.py", line 1778, in _model_default
    'MNE_COREG_PREPARE_BEM', 'true') == 'true')
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 3510, in _init_trait_listeners
    getattr( self, '_init_trait_%s_listener' % data[0] )( name, *data )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 3544, in _init_trait_property_listener
    self.on_trait_change( pre_notify, pattern, priority = True, target=self )
  File "/autofs/cluster/matlab/arch/CentOS5-x86_64/packages/python/anaconda/lib/python2.7/site-packages/traits/has_traits.py", line 2808, in on_trait_change
    listener.set( handler         = ListenerHandler( handler ),
AttributeError: 'NoneType' object has no attribute 'set'

agramfort, 2018-09-13:
ok I can replicate with my old python 2.7 env

can you try with the environment we now recommend?

http://martinos.org/mne/stable/install_mne_python.html#get-mne-and-its-dependencies

larsoner, 2020-04-27:
Closing under the assumption that this is an old traits + Python 2.7 bug, let's reopen if someone hits it again

closed 2020-04-27 by larsoner