mne-tools/mne-python #5529: add_annotation shows only one label

dnacombo, 2018-09-17:
On latest version, running this

```Python
from surfer import Brain

import mne

subjects_dir = mne.datasets.sample.data_path() + '/subjects'
mne.datasets.fetch_hcp_mmp_parcellation(subjects_dir=subjects_dir,
                                        verbose=True)
labels = mne.read_labels_from_annot(
    'fsaverage', 'HCPMMP1', 'lh', subjects_dir=subjects_dir)

brain = Brain('fsaverage', 'lh', 'inflated', subjects_dir=subjects_dir,
              cortex='low_contrast', background='white', size=(800, 600))
brain.add_annotation('aparc.a2009s')
```
shows only one label on the medial wall.

Perhaps it has something to do with the fact that the example below (almost same code) suddenly changed in the current 0.17dev version. On my computer, it shows this white brain with one red label.
http://mne-tools.github.io/dev/auto_examples/visualization/plot_parcellation.html
But I guess it should show this:
https://martinos.org/mne/dev/auto_examples/visualization/plot_parcellation.html

agramfort, 2018-09-17:
FYI it's broken for me too.

larsoner, 2018-09-18:
@dnacombo do you have time to get bisect, or at least check what the last tagged version that works is?

dnacombo, 2018-09-18:
Quick test: white plot_parcellation remains on my install with tagged versions back to 0.14.
0.13 crashes because fetch_hcp_mmp_parcellation does not exist.
But online examples:
http://mne-tools.github.io/0.15/auto_examples/visualization/plot_parcellation.html
is fine
http://mne-tools.github.io/stable/auto_examples/visualization/plot_parcellation.html
is not.
Something odd in the environment, I guess... I can help further but need directions.

larsoner, 2018-09-18:
I mean PySurfer versions not MNE actually. The MNE version shouldn't change
the add_annotation behavior (beyond making the annot dataset available) as
add_annotation is a PySurfer function.

>

agramfort, 2018-09-19:
FYI to make pysurfer plot_parcellation.py I had work with a new env that I
one still
use currently. So it means that maybe updating the env will fix the pb?

dnacombo, 2018-09-20:
Just tried all sorts of variants of `conda update env` without a chance.
`pip install pysurfer` or update or anything won't work.
What should I do?

larsoner, 2018-09-20:
Can you see if it's a `nibabel` problem:

https://github.com/nipy/PySurfer/issues/250#issuecomment-423144386

Let's move discussion over there

closed 2018-09-20 by larsoner