mne-tools/mne-python #5539: raw.plot does not work on my data

erfanpak, 2018-09-20:
I've faced a problem regarding the raw.plot().
I have a .csv file with 7782 X 21 (n_times, n_channel) structure - myData. I've used mne.io.RawArray function to convert my data to a raw data format readable by mne data structure. 
<raw = mne.io.RawArray(myData, info)>
When I am using the raw.plot() function, it does not show me what it's supposed to show as channel activities, although eeglab could show me all the activities including noises (plz take a look at the following pictures). I was wondering where does the problem come from and how I can tackle it? 
Besides, how does the  mne.io.RawArray generally work? 

[image]
[image]
[image]

larsoner, 2018-09-20:
You need to change the scaling of the plot or your data. Probably you have data in units of uV and MNE expects data in V

erfanpak, 2018-09-20:
> You need to change the scaling of the plot or your data. Probably you have data in units of uV and MNE expects data in V

Thanks for the quick response. It is not an **interactive** plot. How can I make it interactive?

agramfort, 2018-09-20:
it's interactive if you don't use %matplotlib inline
in the notebook. You can use "%matplotlib qt" or try
"%matplotlib notebook"

mmagnuski, 2018-10-05:
I'd recommend `%matplotlib` or `%matplotlib qt`, `%matplotlib notebook` doesn't work so well on my machine - it seems to be slower and usually the default figure size does not fit in the output window of the notebook so the navigation is inconvenient.

massich, 2018-10-10:
Closing as solved, @erfpak7 feel free to reopen if you still have troubles

closed 2018-10-10 by massich

shivam-199, 2024-05-09:
It might help input the correct type of data into the array info. For my case, I was analyzing LFP data and set the type as "seeg" (could be wrong). But by doing this, I could get the data to plot without scaling it.

`lfp_info = mne.create_info(nChan, ch_types=["seeg"] * nChan, sfreq=sampling_freq)`