mne-tools/mne-python #5540: compute_source_psd_epochs need 'noisenorm' from inverse operator

alexrockhill, 2018-09-20:
import mne
import os.path as op
from mne.datasets import sample
from mne.minimum_norm import make_inverse_operator, apply_inverse,compute_source_psd_epochs

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'

# Read data
data_path = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
raw = mne.io.read_raw_fif(
    op.join(data_path, 'sample_audvis_raw.fif'), preload=True)
raw.load_data().filter(None, 9, fir_design='firwin')
raw.set_eeg_reference('average', projection=True)  # set EEG average reference
event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4, 'smiley': 5, 'button': 32}
events = mne.read_events(op.join(data_path, 'sample_audvis_raw-eve.fif'))
epochs = mne.Epochs(raw, events, event_id=event_id, tmin=-0.2, tmax=.5)
fname_fwd = sample.data_path() + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'
fname_cov = sample.data_path() + '/MEG/sample/sample_audvis-cov.fif'
fwd = mne.read_forward_solution(fname_fwd)
cov = mne.read_cov(fname_cov)

inv = make_inverse_operator(epochs.info, fwd, cov, loose=0., depth=0.8,
                            verbose=True)

epochs = epochs.load_data()
stc = compute_source_psd_epochs(epochs, inv, lambda2=1.0, method='dSPM', fmin=7.0, fmax=35.0, pick_ori='normal', label=None, nave=len(epochs), pca=True, inv_split=None, bandwidth=4.0, adaptive=False, low_bias=True, return_generator=False, n_jobs=1, prepared=True, method_params=None, verbose=None)

alexrockhill, 2018-09-20:
The issue was with preload = True in the keyword arguments.

larsoner, 2018-09-20:
To clarify for future folks, I think you mean `prepared=True` here.

I'm going to reopen so we remember to catch non-prepared `inv` use in these sorts of functions

PR #5702 (2018-11-08): MRG: Fix check for prepared
Closes #5540.
closed 2018-11-08 by larsoner