mne-tools/mne-python #5550: AUX channel type

Sonia2016, 2018-09-23:
Hi all,
I am using **B-Alert X24 EEG System** and it has 24 channels which 20 of them are EEG type and one of them is ECG type but I do not know the types of rest of them which are `'AUX1 (Raw)', 'AUX2 (Raw)', 'AUX3 (Raw)'`. Do you know by any chance what can be the type of `AUX`? 
Thanks!

dasdiptyajit, 2018-09-24:
@Sonia2016 can you read the raw file and type?:  raw.info['ch_names']. It should provide you with all the channel names.

Sonia2016, 2018-09-24:
> @Sonia2016 can you read the raw file and type?: raw.info['ch_names']. It should provide you with all the channel names. I guess those are stim/resp channels.

Thanks @dasdiptyajit ! Unfortunately, I do not have types of my channels in raw.info. The reason is that I've created the raw.info by myself based on the .txt data created by B-Alert. I am not quite sure what do you mean by _stim/resp channels_, are the types of channels if yes how should I name it in the raw.info?

agramfort, 2018-09-25:
you can pick any name you like. It's just the channel type that will matter
as it is liked to physical units like Volts that are used for plotting.

dasdiptyajit, 2018-09-25:
@Sonia2016 'AUX1 (Raw)', 'AUX2 (Raw)', 'AUX3 (Raw)' could be auxiliary channels (e.g., typically ECG/EOG channels) or other external trigger channels such as stimulus or response. You should try to look at the technical specification of your EEG system. Anyhow, the naming/tagging doesn't matter much as long as you use the channels in a right way with proper physical units that @agramfort has already mentioned.

Sonia2016, 2018-09-25:
> @Sonia2016 'AUX1 (Raw)', 'AUX2 (Raw)', 'AUX3 (Raw)' could be auxiliary channels (e.g., typically ECG/EOG channels) or other external trigger channels such as stimulus or response. You should try to [...]

Thanks so much, @dasdiptyajit ! I Placed the mastoid channels (AUX1-AUX2) directly on the bone behind right and left ears. As well as two channels (AUX3 and AUX4) on collar bone which I think one of them has automatically been replaced by ECG. Can I still consider AUX1 and two as EOG considering the fact that they are both attached behind the ears?

agramfort, 2018-09-25:
use the set_channel_types method of raw to set properly your channel types.
If your pb is solved please close the issue

dasdiptyajit, 2018-09-27:
@Sonia2016 I would additionally select one frontal EEG channel (e.g., Fp1/Fp2) to use it as EOG channel since the sensor located close to the eyes has more chances to pick up eye blinks.

Sonia2016, 2018-09-30:
Thanks @dasdiptyajit! that was very helpful!