mne-tools/mne-python #5552: ENH: Convienance methods for labels

bloyl, 2018-09-24:
It would be good if `stc.extract_label_time_course` worked on stcs that are already confined to labels

Similarly it would be useful to have a method for putting derived measures back in to stc form for visualization.

Below is a use-case although more meaning full uses would come from `apply_inverse_raw` and `apply_inverse_epochs`. 

I don't know my way around labels and sourcespaces, but if someone would like to give me a quick break down on how these could be resolved I'd be happy to take a shot at it.

```
import mne
from mne.datasets import sample
from mne.minimum_norm import read_inverse_operator, apply_inverse

data_path = sample.data_path()
label = 'Aud-lh'
label_fname = data_path + '/MEG/sample/labels/%s.label' % label
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'
fname_evoked = data_path + '/MEG/sample/sample_audvis-ave.fif'
subjects_dir = data_path + '/subjects'

snr = 3.0
lambda2 = 1.0 / snr ** 2
method = "dSPM"  # use dSPM method (could also be MNE or sLORETA)

# Load data
evoked = mne.read_evokeds(fname_evoked, condition=0, baseline=(None, 0))
inverse_operator = read_inverse_operator(fname_inv)
src = inverse_operator['src']

label = mne.read_label(label_fname)

# Compute inverse solution
pick_ori = "normal"  # Get signed values to see the effect of sign filp
stc = apply_inverse(evoked, inverse_operator, lambda2, method,
                      pick_ori=pick_ori, label=label)

stc_full = apply_inverse(evoked, inverse_operator, lambda2, method,
                         pick_ori=pick_ori)

stc_in_label = stc_full.in_label(label)

# these fails!
mean = stc.extract_label_time_course(label, src, mode='mean')
mean_2 = stc_in_label.extract_label_time_course(label, src, mode='mean')

# it would then be good to be able to put it (or some derived measure) back
stc_label = mne.label_data_to_stc(label, mean, src)

```

agramfort, 2018-09-25:
if label.hemi == 'lh' basically you need to find the intersection between label.vertices and src[0]['vertno'] then the matching vertices give you the vertices[0] to be passed to the SourceEstimate constructor.

you do the same for right hemi 'rh' and replace 0 by 1.

makes sense?

bloyl, 2018-09-26:
Are there any labels based on volume source spaces that I need to worry about?

agramfort, 2018-09-26:
not AFAIK or at least I never used such. For sure they don't have an
hemi attribute

larsoner, 2019-09-19:
> It would be good if stc.extract_label_time_course worked on stcs that are already confined to labels

Agreed these should work and give the same result (I'll look into it):
```
stc = apply_inverse(evoked, inverse_operator, lambda2, method,
                      pick_ori=pick_ori, label=label)
stc_full = apply_inverse(evoked, inverse_operator, lambda2, method,
                         pick_ori=pick_ori)
stc_in_label = stc_full.in_label(label)
mean = stc.extract_label_time_course(label, src, mode='mean')
mean_2 = stc_in_label.extract_label_time_course(label, src, mode='mean')
```

> Similarly it would be useful to have a method for putting derived measures back in to stc form for visualization.

We have this now with http://mne.tools/dev/generated/mne.labels_to_stc.html

PR #6789 (2019-09-21): MRG, BUG: Fix behavior with stc.in_label
Closes #5552.
closed 2019-09-21 by agramfort