mne-tools/mne-python #5553: Localizing electrodes using a 3D-scanner

mkoculak, 2018-09-25:
Hi,
first of all, thank you for a great tool!

Recently we have bought a 3D scanner for digitalization of the actual electrode location on subject's head.
You can see the scanner and the usage in the fieldtrip tutorial below:
http://www.fieldtriptoolbox.org/tutorial/3dscanner

I was wondering is there a way right now to incorporate those 3D scans in MNE analysis?
Or would it require coding the functionality from scratch?

Maybe it would be a good idea for an enhancement, if this is impossible right now in python.
Thanks in advance for any suggestions!

agramfort, 2018-09-26:
I don't know anyone who has done this.

mayavi supports 3d picking of vertices so technically all the pieces exist
but it remains to be assembled...

mmagnuski, 2018-10-04:
I am using photogrammetry to digitize channels too but I am using a video from a normal handheld camera and then reconstruct 3d object and channel positions in 3DF Zephyr (I will try with cellphone camera in a few month to see if that can be used). However, I export the model in the same format that fieldtrip uses (wavefront .obj). I tried using fieldtrip functions to read the zephyr files but it seems that matlab is not able to texture the meshes with enough detail (it cannot apply a texture patch to a triangle - only a single color to one triangle or edge/vertex is possible). I am not sure if mayavi can do it better but I am sure it should be possible to at least replicate what fieldtrip does in mne.
It might be worth trying it first in a standalone repo and adding it later to mne if it would be useful to the mne community. There are some python packages for reading wavefront .obj files and we could assume, as fieldtrip does, that only one texture file is used for the model. I never did any 3d picking in mayavi but with some help I could try that too. :)

mmagnuski, 2018-10-05:
Ok, it seems that mayavi can texture each triangle given the texture image and vertex coordinates on the texture. It was not so easy to find that information online so I expect that we may encounter some difficulities along the road:
http://enthought-dev.117412.n3.nabble.com/Add-texture-coordinates-to-triangular-mesh-td763116.html

@agramfort - where could I find some good intro to interactive mayavi (especially handling clicks on a 3d object)?

larsoner, 2018-10-05:
Look in the coreg GUI code there is interactive clicking to set the MRI fiducials

agramfort, 2018-10-05:
you can generally press p in a vtk figure to pick a vertex.

sappelhoff, 2021-01-15:
Note that this method (based on Robert and Simon Homölle's https://doi.org/10.1016/j.jneumeth.2019.108378) is also implemented in EEGLAB:

- https://github.com/sccn/get_chanlocs/wiki
- there is also a slideshow: https://sccn.ucsd.edu/mediawiki/images/5/5f/Get_chanlocs_userguide.pdf

mmagnuski, 2021-01-22:
see also: https://pubmed.ncbi.nlm.nih.gov/28559791/ for an approach that relies on normal images (not using structured light scanner) and seems to auto-detect EEG channels. The GUI is also implemented in MATLAB.