mne-tools/mne-python #5561: ENH: Speed up time frequency code

larsoner, 2018-09-27:
I adapted @jona-sassenhagen's example to make it smaller:
```
import numpy as np
import time
import mne

sfreq = 128.
n_epochs, n_ch, n_time = 144, 29, 1921
data = np.zeros((n_epochs, n_ch, n_time))
info = mne.create_info(n_ch, sfreq, 'eeg')
epochs = mne.EpochsArray(data, info)

t0 = time.time()
mne.time_frequency.tfr_morlet(epochs, [1, 2], n_cycles=3, use_fft=True,
                              n_jobs='cuda', verbose=True)
print(time.time() - t0)
```
Using CUDA made it slower. I think we should:

1. Refactor our overlap-add filtering code to allow multiple filters applied to the same data (we do this in CWT).
2. Refactor our overlap-add filtering code to work with complex data (morlet coefs).
3. refactor _cwt (the core of most TFR functions) to use overlap-add filtering. That should be faster on CPU already.
4. deprecate `use_fft` (it should always be faster)
5. Add support for CUDA (should be trivial after the `_cwt` refactoring)

jona-sassenhagen, 2018-09-28:
I like that you say "we".

>deprecate use_fft (it should always be faster)

Oh, that sounds like a no brainer. Fewer options is better.

larsoner, 2020-04-27:
I think with faster FFTs available via pocketfft in SciPy and NumPy nowadays this is probably less critical. Though perhaps we should someday change the default `use_fft=True` since it's probably always faster nowadays.

closed 2020-04-27 by larsoner