mne-tools/mne-python #5563: Error with make_inverse_operator

pripolles, 2018-09-27:
Hi! We are trying to run source analysis.  @kingjr 
We are running MNE version '0.16.dev0'

So far this is what we have done:

1 We prepare the source space:
```
src = setup_source_space(subject=freesurfer_subject, subjects_dir=freesurfer_dir,
                                 spacing='oct6', surface='white',
                                 add_dist=True, n_jobs=-1, verbose=None)
bem_dir = os.path.join(freesurfer_dir, freesurfer_subject, 'bem')
src_fname = os.path.join(bem_dir, freesurfer_subject + '-oct-6-src.fif')
write_source_spaces(src_fname, src, overwrite=True)
```
2 We run the forward model 
```
fwd = forward_pipeline(raw_fname, subjects_dir=freesurfer_dir, subject=freesurfer_subject,
                       overwrite=overwrite)
```

3 We compute noise covariance
```
covs = compute_covariance(epochs, tmin=None, tmax=0,
                          method=['shrunk', 'empirical'],
                         return_estimators=True)
covs[0].save(cov_fname)
cov = covs[0]
```

4 And then, when we try to calculate the inverse model:
`inv = make_inverse_operator(epochs.info, fwd, cov, loose=0.2, depth=0.8)`

We get this error:

> TypeError: 'NoneType' object has no attribute '__getitem__'

Here you have a caption with the full report:
[image]

Does anybody know what might be wrong here? Many thanks in advance

larsoner, 2018-09-27:
`forward_pipeline` is not an MNE function so it's tough to help you debug. But based on the error it looks like `forward[...]` is causing the problem, which means the `fwd` you have is probably `None`

larsoner, 2018-11-07:
Closing due to non-response / likely bug in custom code, but if you can isolate it to MNE calls let us know

closed 2018-11-07 by larsoner