mne-tools/mne-python #5567: plot channel over epochs

Sonia2016, 2018-09-30:
I'm trying to visualize channel over epochs or event related potential image. I've found the following [https://martinos.org/mne/stable/auto_examples/visualization/plot_channel_epochs_image.html#sphx-glr-auto-examples-visualization-plot-channel-epochs-image-py](url).  I'd like to get a similar nice looking figure. However, the figure turns out like what I've attached which is not even close to what I was hoping to get. I do know how to fix it or what I'm doing wrong. Do you know how can we change the range?
One hypothesis (I might be so wrong) is that, as can be seen in `mne.viz.plot_epochs_image` the output in the heatmap and the time series figure have the same range of _ft/cm_ which in my case  these two ranges are different and eventually the heatmap value doesn't make sense and it is not mapping the time series value as you can check in this image. 

[image]

jona-sassenhagen, 2018-10-01:
Can you show the code you used?

Sonia2016, 2018-10-01:
As I mentioned, I've used the following code
```
good_pick = 19  # channel with a clear evoked response
bad_pick = 6  # channel with no evoked response

# We'll also plot a sample time onset for each trial
plt_times = np.linspace(0, .2, 62)#len(epochs))

plt.close('all')
mne.viz.plot_epochs_image(epochs["event_id"], [good_pick, bad_pick], sigma=.05,
                          order=order_func, vmin=-250, vmax=250, 
                          overlay_times=plt_times, show=True)`
```

In order to get a grasp of my dataset, it contains 24 channels (_ABM-BAlet 24_) including one ecg and three AUX channels with a sfreq of 256Hz. I recorded the EEG file through iMotions software which provides a .txt file. For the purpose of MNE analysis, I manually converted it to a MNE data format with `mne.io.RawArray(myData, info)`. **_myData_ shape is (21,436224) (n_channel, n_rows_of_data)**. I've defined the info module as follow ` info = mne.create_info( ch_names=['ECG', 'Fp1', 'F7', 'F8', 'T4', 'T6', 'T5', 'T3', 'Fp2', 'O1', 'P3', 'Pz', 'F3', 'Fz', 'F4', 'C4', 'P4', 'POz', 'C3', 'Cz', 'O2'], ch_types=['ecg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg', 'eeg'], sfreq=256, montage=montage )`. 
**Not to mention that, I assumed that each epoch represents n_row_of_data (436224) divided by sfreq (256)** (is that right?) 
As for the events, the events were derived from event_id = dict(liveMarker=1,IGNORED=0). Turns out that the _liveMarker_ includes  62 actual events and _IGNORED_ happened to be repeated 436162 times in my data set. 
Them, I created the epochs as follow e
`pochs_params = dict(events=events, event_id=event_id, tmin=-0.1,tmax = 0.55, baseline=(-0.1,0),preload= True) epochs = mne.Epochs(raw, **epochs_params)`
 I hope those all make sense. If does not, I’d happy to share further information to make it crystal clear.
 
Back to the initial question, as all the taken steps, I’m not quite sure how to fix the ERP plot to make it here: [https://martinos.org/mne/stable/auto_examples/visualization/plot_channel_epochs_image.html#sphx-glr-auto-examples-visualization-plot-channel-epochs-image-py](url)
I’ll be so grateful if you could explain me in details what I’ve missed and how should I correct them. Do I Need to do any preprocessing before doing ERP? Is there any problem with the scale of uV which is completely different as can be seen in the previous msg?

agramfort, 2018-10-01:
@Sonia2016 are your data scaled in Volts raw object? What you access data in raw you should see values in the order of a few hundreds of uV so in the 1e-4 range. I fear your data are stored in uV

jona-sassenhagen, 2018-10-01:
You’re manually specifying vmin and vmax (the values from the example are for MEG data). Remove these parameters and see what the defaults do for you.

agramfort, 2018-10-01:
no scaling is done wrt channel type. Here it says values are between -250uV
and 250uV

Sonia2016, 2018-10-01:
Thanks, @agramfort! I think I didn't totally understand what do you mean by scaling with regard to channel type, could you please let me know how can I do that? 
How can I check the scale of data (uV or V)? 

@jona-sassenhagen  the default works! you can check in the image. but the value (range) of y-axis of time series is not the same as heatmap color scale, do you know why? how can I change those?
[image]

@agramfort , @jona-sassenhagen how can I change the scale of data?

Sonia2016, 2018-10-01:
Thanks @jona-sassenhagen!
@agramfort, just want to check with you, this is one part of my data set, I've highlighted the value for each channel here. Do you know how can I get the scale of data (for each channel)?
[image]

agramfort, 2018-10-01:
@Sonia2016 just look at the plot. The values are in the order of 1e7 uV so you created a raw objects from uV and not V. Just multiply the values by 1e-6 when creating your raw object.

please close the issue if it's fixes the pb.

jona-sassenhagen, 2018-10-03:
@Sonia2016 can we close?

Sonia2016, 2018-10-03:
I will! :)

closed 2018-10-03 by Sonia2016