mne-tools/mne-python #5579: How to compute inverse operator for source localization

vivekosdd, 2018-10-04:
I'm working on source localization with MEG data. I'm using the following tutorial for source localization.
https://mne-tools.github.io/0.12/auto_tutorials/plot_mne_dspm_source_localization.html

I cannot compute the inverse operator. The error I'm getting is as follows:
`>>> inverse_operator = make_inverse_operator(info, fwd, noise_cov, loose=0.2, depth=0.8)
    Average patch normals will be employed in the rotation to the local surface coordinates....
    Converting to surface-based source orientations...
    [done]
Computing inverse operator with 0 channels.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 2, in make_inverse_operator
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/minimum_norm/inverse.py", line 1471, in make_inverse_operator
    _prepare_forward(forward, info, noise_cov, rank=rank)
  File "<string>", line 2, in _prepare_forward
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/minimum_norm/inverse.py", line 1301, in _prepare_forward
    rank, pca)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/cov.py", line 1272, in _get_whitener
    noise_cov = prepare_noise_cov(noise_cov, info, ch_names, rank)
  File "<string>", line 2, in prepare_noise_cov
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/cov.py", line 1342, in prepare_noise_cov
    ch_names)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/io/proj.py", line 548, in make_projector
    return _make_projector(projs, ch_names, bads, include_active)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/io/proj.py", line 561, in _make_projector
    raise ValueError('No channel names specified')
ValueError: No channel names specified`

It says No channel names specified whereas I have channel names info in the file.
Please help.
My data is a resting state data. Please let me know if there is any tutorial for source localization of resting state data.
Thank You

vivekosdd, 2018-10-04:
No, the error is with my data not the tutorial.

agramfort, 2018-10-04:
do you reuse some of the files from the tutorial? if so I am pretty
sure you should not.

vivekosdd, 2018-10-04:
I only use the forward solution file as a template

agramfort, 2018-10-05:
@vivekosdd you cannot do this. You need individual sensor locations. You can use a template anatomy by scaling fsaverage with the coreg GUI but ideally you have the T1 MRI of your subject.

closed 2018-10-05 by agramfort

agramfort, 2020-06-11:
@barlehmann this issue is closed. don't use it to ask usage questions. Use the MNE mailing list or the gitter channel https://gitter.im/mne-tools/mne-python

barlehmann, 2020-06-14:
Certainly, will do, sorry about that.