mne-tools/mne-python #5592: Warnings are displayed twice

cbrnr, 2018-10-08:
Whenever a warning is emitted, the message is shown twice because the logger also prints this information to the screen. This results into warnings that are difficult to read (e.g. it is hard to tell how many warnings occurred). For example:

```
Extracting parameters from /Users/clemens/Desktop/Functions_1b_16.vhdr...
<ipython-input-2-ea8161bea387>:1: RuntimeWarning: No coordinate information found for channels ['EXG1', 'EXG2', 'EXG3', 'EXG4', 'EXG5', 'EXG6', 'EXG7', 'EXG8', 'Status']. Setting channel types to misc. To avoid this warning, set channel types explicitly.
  raw = mne.io.read_raw_brainvision("/Users/clemens/Desktop/Functions_1b_16.vhdr", preload=True)
No coordinate information found for channels ['EXG1', 'EXG2', 'EXG3', 'EXG4', 'EXG5', 'EXG6', 'EXG7', 'EXG8', 'Status']. Setting channel types to misc. To avoid this warning, set channel types explicitly.
Setting channel info structure...
Reading 0 ... 100863  =      0.000 ...   393.996 secs...
Used Annotations descriptions: ['Stimulus/S 11', 'Stimulus/S120', 'Stimulus/S  2', 'Stimulus/S 90', 'Stimulus/S  1', 'Stimulus/S 13', 'Stimulus/S 70', 'Stimulus/S 77', 'Stimulus/S 86', 'Stimulus/S119', 'Stimulus/S112', 'Stimulus/S 67', 'Stimulus/S 96', 'Stimulus/S 75', 'Stimulus/S 83', 'Stimulus/S118', 'Stimulus/S 68', 'Stimulus/S113', 'Stimulus/S100', 'Stimulus/S 74', 'Stimulus/S 93', 'Stimulus/S 72', 'Stimulus/S 84', 'Stimulus/S 64', 'Stimulus/S108', 'Stimulus/S 85', 'Stimulus/S106', 'Stimulus/S 81', 'Stimulus/S 79', 'Stimulus/S 73', 'Stimulus/S 89', 'Stimulus/S111', 'Stimulus/S 61', 'Stimulus/S 71', 'Stimulus/S115', 'Stimulus/S 82', 'Stimulus/S 78', 'Stimulus/S 65', 'Stimulus/S109', 'Stimulus/S 66', 'Stimulus/S 98', 'Stimulus/S104', 'Stimulus/S 62', 'Stimulus/S117', 'Stimulus/S 69', 'Stimulus/S 91', 'Stimulus/S 97', 'Stimulus/S101', 'Stimulus/S 80', 'Stimulus/S 63', 'Stimulus/S 87', 'Stimulus/S 92', 'Stimulus/S105', 'Stimulus/S 99', 'Stimulus/S116', 'Stimulus/S102', 'Stimulus/S110', 'Stimulus/S103', 'Stimulus/S 88', 'Stimulus/S 94', 'Stimulus/S114', 'Stimulus/S 95', 'Stimulus/S107', 'Stimulus/S 76']
<ipython-input-2-ea8161bea387>:1: RuntimeWarning: 7 annotation(s) will be dropped, such as ['Comment/ActiveTwo MK2', 'Comment/Start Epoch', 'Comment/CMS in range', 'Comment/Speed Mode 4', 'Bad Interval/Multiple']. Consider using ``regexp`` to ignore annotations that do not follow a specific pattern.
  raw = mne.io.read_raw_brainvision("/Users/clemens/Desktop/Functions_1b_16.vhdr", preload=True)
7 annotation(s) will be dropped, such as ['Comment/ActiveTwo MK2', 'Comment/Start Epoch', 'Comment/CMS in range', 'Comment/Speed Mode 4', 'Bad Interval/Multiple']. Consider using ``regexp`` to ignore annotations that do not follow a specific pattern.
```

Ideally, warnings should be prominently visible, but only once. I could completely disable the logger, but then I wouldn't be able to see messages with less severity than warnings (i.e. infos).

agramfort, 2018-10-08:
fair comment. How do you suggest to proceed?

cbrnr, 2018-10-09:
No idea. How do other projects handle this? We could hide logging output for warnings/errors, but I'm pretty sure this is not a good solution. We could also write all logging messages to a file, which is probably also not ideal. However, if we clutter the output people will likely just ignore all messages (including really important ones).

agramfort, 2018-10-09:
let's keep thinking then but I agree with the concern.

christianbrodbeck, 2018-10-09:
What is the reason for adding a module-specific handler that writes to stdout? Wouldn't this issue go away by not doing that?

larsoner, 2018-10-09:
> What is the reason for adding a module-specific handler that writes to stdout? Wouldn't this issue go away by not doing that?

I'm not sure what you mean. We do some module-related magic to figure out the `stacklevel` in a `warnings.warn` call but otherwise I don't think we do anything module-specific. We basically do a `warnings.warn(..., stacklevel=...)` and `logger.warning(...)`. The former is useful for seeing the stack level and parsing warnings / stderr, the latter is useful in a logging context, which for example you can have write to a file (`mne.set_log_file`).

Do you mean something else?

christianbrodbeck, 2018-10-09:
In the `utils` module, a new handler is added upon initialization (https://github.com/mne-tools/mne-python/blob/master/mne/__init__.py#L118). The reason the warnings are printed twice is that they are printed once by the standard Python handler (which has level WARNINGS unless changed) and then by the `mne` handler.

larsoner, 2018-10-09:
> The reason the warnings are printed twice is that they are printed once by the standard Python handler (which has level WARNINGS unless changed) and then by the mne handler. Or did I misunderstand the issue?

This is not why it happens. In `utils.py` we have:
```
        warnings.warn_explicit(message, category, fname, lineno,
                               'mne', globals().get('__warningregistry__', {}))
    logger.warning(message)
```
If you comment out either of those two lines, you only get one message printed.

larsoner, 2018-10-09:
... which actually makes me realize one workaround/solution is to check if any of the `logger` handlers is a file handler. If so, send the message (and you'll get two in stdout, which is less than ideal). If not (which is the far more common use case) don't.

christianbrodbeck, 2018-10-09:
Aah ok, sorry for my confusion then.

PR #5596 (2018-10-10): MRG: Only output one warning
Closes #5592.

@cbrnr can you verify it fixes your issue?

The changes in `test_maxwell.py` are from when I started to modify the file to accommodate these changes before I realized there was a simpler solution (monkey-patching the `catch_logging` handler), but the changes ended up being nice anyway so I decided to leave them in.
closed 2018-10-10 by agramfort