mne-tools/mne-python #5599: unify treatment of deprecation warnings in the tests

massich, 2018-10-10:
Based on https://github.com/mne-tools/mne-python/pull/5580/#discussion_r223900833 discussion.

The contribution guide https://mne-tools.github.io/stable/contributing.html?highlight=contribute recommends how to deprecate a functionality but not how to test it. Therefore, we find `pytest.deprecated_call` and ` pytest.warns(DeprecationWarning` in our code base.

```sh
~/code/mne-python _read_annot*
(mne27) ❯ git grep "pytest.depre"       
mne/tests/test_morph.py:    with pytest.deprecated_call():
mne/tests/test_morph.py:    with pytest.deprecated_call():
mne/tests/test_morph.py:    with pytest.deprecated_call():
mne/tests/test_utils.py:    pytest.deprecated_call(deprecated_func)
mne/tests/test_utils.py:    pytest.deprecated_call(deprecated_class)

~/code/mne-python _read_annot*
(mne27) ❯ git grep "pytest.warns(DeprecationWarning"
mne/io/brainvision/tests/test_brainvision.py:    with pytest.warns(DeprecationWarning):
mne/io/brainvision/tests/test_brainvision.py:    with pytest.warns(DeprecationWarning):
mne/io/edf/tests/test_edf.py:    with pytest.warns(DeprecationWarning):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/eeglab/tests/test_eeglab.py:    with pytest.warns(DeprecationWarning, match="read_events_eeglab"):
mne/io/tests/test_raw.py:    with pytest.warns(DeprecationWarning, match='by assignment is deprecated'):
mne/io/tests/test_raw.py:    with pytest.warns(DeprecationWarning):
mne/io/tests/test_raw.py:    with pytest.warns(DeprecationWarning):
```
we should unify it and document it.

massich, 2018-10-10:
for completeness

```py
from mne.utils import deprecated
from mne.utils import warn

@deprecated('my_function will be deprecated in 0.XX, please use my_new_function instead.')
def my_deprecated_function():
   return 'foo'

def my_deprecated_param(new_param=None, old_param=None):
    if old_param is not None:
         warn('old_param is deprecated and will be replaced by new_param in 0.XX.',
              DeprecationWarning)
         new_param = old_param
    # Do what you have to do with new_param
    return 'foo'

def my_not_deprecated_function():
    pass

def test_treat_deprecation_warnings_A():
    with pytest.warns(DeprecationWarning, match="my_new_function"):
        my_deprecated_function()
    with pytest.warns(DeprecationWarning, match="old_param .* new_param"):
        my_deprecated_param(old_param='bar')
    with pytest.warns(DeprecationWarning, match="is not here"):
        my_deprecated_function()

def test_treat_deprecation_warnings_B():
    with pytest.deprecated_call(match="my_new_function"):
        my_deprecated_function()
    with pytest.deprecated_call(match="old_param .* new_param"):
        my_deprecated_param(old_param='bar')
    with pytest.deprecated_call(match="is not here"):
        my_deprecated_function()
    with pytest.deprecated_call():
        my_not_deprecated_function()
```

massich, 2018-10-10:
I totally agree that's why open an issue at pytest.

massich, 2018-10-10:
From now on, I'll write `pytest.deprecated_call` (with match=`xx` even if its not working yet) and let this issue for a newcomer.

larsoner, 2018-10-10:
Sounds good

massich, 2018-10-10:
I tried to find the `good first issue` label, etc.. those that github recommend, but they were not there and I decided that I would do it at some other point.

larsoner, 2018-10-10:
We just use the EASY label for that

larsoner, 2019-09-21:
Sounds like `deprecated_call` is what we can/should try to stick to, and we can do this for 0.20 deprecations so I'll close this in slight anticipation of doing so

closed 2019-09-21 by larsoner