mne-tools/mne-python #5600: mne coreg issue

dasdiptyajit, 2018-10-10:
When running with 0.17.dev0: mne coreg,  I got the following error:

**Error:**
Triangle neighbors and vertex normals...
Copying fsaverage subject from freesurfer directory...
Exception occurred in traits notification handler.
Please check the log file for details.
Exception occurred in traits notification handler for object: <mne.gui._file_traits.SubjectSelectorPanel object at 0x7ff718ef2780>, trait: create_fsaverage, old value: <undefined>, new value: 0
Traceback (most recent call last):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_file_traits.py", line 562, in _create_fsaverage_fired
    self.model.create_fsaverage()
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_file_traits.py", line 517, in create_fsaverage
    subjects_dir=self.subjects_dir)
  File "<string>", line 2, in create_default_subject
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/coreg.py", line 193, in create_default_subject
    os.mkdir(dest_bem)
PermissionError: [Errno 13] Permission denied: '/media/diptyajit/Ext_drive/FreeSurfer_Home/fsaverage/bem'
Fatal Python error: Segmentation fault

Thread 0x00007ff7241e9700 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 295 in wait
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/queue.py", line 164 in get
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_coreg_gui.py", line 1345 in worker
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 864 in run
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007ff76ae94740 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 568 in _wrap_call
  File "tvtk_classes/algorithm.py", line 800 in update
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/filters/filter_base.py", line 87 in update_data
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/sources/vtk_data_source.py", line 320 in _fire_data_changed
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 250 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 310 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 568 in _wrap_call
  File "tvtk_classes/point_set.py", line 57 in _set_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_viewer.py", line 195 in _update_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_viewer.py", line 538 in _update_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_coreg_gui.py", line 2029 in _update_mri_obj
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 3371 in notify
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 965 in wrapper0
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 552 in _notify_function_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 3371 in notify
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 965 in wrapper0
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 552 in _notify_function_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_file_traits.py", line 188 in read_file
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
[... 27 lines cut ...]
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/utils.py", line 729 in verbose
  File "<string>", line 2 in coregistration
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/commands/mne_coreg.py", line 110 in run
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/commands/mne_coreg.py", line 117 in <module>

**output of mne.sys_info():**
Platform:      Linux-4.15.0-33-generic-x86_64-with-debian-buster-sid
Python:        3.6.6 |Anaconda, Inc.| (default, Oct  9 2018, 12:34:16)  [GCC 7.3.0]
Executable:    /home/diptyajit/anaconda3/envs/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        15.6 GB

mne:           0.17.dev0
numpy:         1.15.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=Qt5Agg}

sklearn:       0.20.0
nibabel:       2.3.0
mayavi:        4.6.2 {qt_api=pyqt5}
cupy:          Not found
pandas:        0.23.4
dipy:          0.14.0
 
Is it a issue from mayavi end? Is there any way to upgrade the mayavi (dev version) under mne environment or any other possible solution? thanks!

larsoner, 2018-10-10:
The segfault is a bummer, you can try
```
pip install --upgrade "https://api.github.com/repos/enthought/mayavi/zipball/master"
```

But it looks like there is also a problem where you don't have permissions to modify the `SUBJECTS_DIR`. You should probably use a different directory where you have write permissions.

dasdiptyajit, 2018-10-10:
@larsoner  I have updated mayavi (i.e., 4.7.0.dev0)  and resolved the write permissions. Now I have the following error:

/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
    Triangle neighbors and vertex normals...
    Triangle neighbors and vertex normals...
Fatal Python error: Segmentation fault

Thread 0x00007fcc478e6700 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 295 in wait
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/queue.py", line 164 in get
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/gui/_coreg_gui.py", line 1345 in worker
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 864 in run
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007fcca809d740 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 595 in _wrap_call
  File "tvtk_classes/algorithm.py", line 800 in update
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/filters/filter_base.py", line 87 in update_data
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/sources/vtk_data_source.py", line 321 in _fire_data_changed
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 250 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 310 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 595 in _wrap_call
  File "tvtk_classes/point_set.py", line 57 in _set_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/gui/_viewer.py", line 195 in _update_points
 
Any idea how to solve this?

larsoner, 2018-10-10:
Does this work for you:

from mayavi import mlab
mlab.test_plot3d()

?

dasdiptyajit, 2018-10-11:
@larsoner its a bit strange. But I got different results: 

**mayavi 3d visualizer looks fine**
[image]

**Using python 3.6, the plot looks fine as well **
[image]

However,
**In ipython, the output is frozen and I had to kill the process. This is same with Spyder and with Pycharm I can't even import mlab **
[image]
[image]

agramfort, 2018-10-11:
@prabhuramachandran any idea?

larsoner, 2018-10-11:
In IPython did you try `%gui qt` as suggested [in the install docs](http://mne-tools.github.io/dev/install_mne_python.html#check-that-everything-works)?

dasdiptyajit, 2018-10-11:
yes, it's working now with Ipython.I am not sure if it's a mayavi issue. mne coreg is still not fixed.
 
**Still getting the following error:**

corrupted double-linked list
Fatal Python error: Aborted

Thread 0x00007faeb22e9700 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 295 in wait
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/queue.py", line 164 in get
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_coreg_gui.py", line 1345 in worker
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 864 in run
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007faf32a49740 (most recent call first):
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 595 in _wrap_call
  File "tvtk_classes/algorithm.py", line 800 in update
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/filters/filter_base.py", line 87 in update_data
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mayavi/sources/vtk_data_source.py", line 321 in _fire_data_changed
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 250 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/messenger.py", line 310 in send
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/tvtk/tvtk_base.py", line 595 in _wrap_call
  File "tvtk_classes/point_set.py", line 57 in _set_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_viewer.py", line 195 in _update_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_viewer.py", line 538 in _update_points
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_coreg_gui.py", line 2029 in _update_mri_obj
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 545 in _notify_method_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 3371 in notify
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 965 in wrapper0
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 552 in _notify_function_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 3371 in notify
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/has_traits.py", line 965 in wrapper0
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 552 in _notify_function_listener
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 474 in __call__
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/_file_traits.py", line 188 in read_file
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 482 in dispatch
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/traits/trait_notifiers.py", line 519 in _dispatch_change_event
[... cut ...]
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/gui/__init__.py", line 171 in coregistration
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/utils.py", line 801 in verbose
  File "<string>", line 2 in coregistration
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/commands/mne_coreg.py", line 110 in run
  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.6/site-packages/mne/commands/mne_coreg.py", line 117 in <module>

prabhuramachandran, 2018-10-11:
@agramfort -- I guess @larsoner has answered the issue the OP was facing with spyder/ipython.  As regards the long error, I have no idea what is wrong.

larsoner, 2018-11-07:
> Still getting the following error: whenever I tried to select a different subject (i.e., mne coreg fails)

I have seen people have this segfault, too. The workaround is to (annoyingly) just close and reopen `mne coreg` from the command line with each new subject.

larsoner, 2019-05-13:
More reports of this are in #5728 and the discussion there is more streamlined, so closing this

closed 2019-05-13 by larsoner