mne-tools/mne-python #5605: Colorbar scale changes intermediate values in stc.plot

alexrockhill, 2018-10-12:
The colorscale between the left, right and middle points changes around per the example:

import mne
data_path = mne.datasets.sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis-meg'
stc = mne.read_source_estimate(fname)
fig = stc.plot(subjects_dir=data_path+'/subjects/',subject='sample',hemi='both',views=['lat'],colormap='mne')
fig.save_movie('test.gif',time_dilation=10)
fig.close()

Now, it seems to look fine in the video when the figure is shown, but when you look at test.gif, the colorbar is changing around a lot.

larsoner, 2018-10-17:
This looks like a gif palette / saving problem. the colorbar "changes" in that which parts get which value differ from frame to frame. The same phenomenon happens in the "old pysurfer" case, it's just less obvious. Look closely and you'll see it.

larsoner, 2018-10-17:
... in other words, it's going to be a problem with `imageio.mimwrite` and/or the `gif` format.

In this case, using `subrectangles=True` works around the problem (since the colorbar is outside the "changing rectangle" from frame to frame)
```
fig.save_movie('test.gif', time_dilation=10, subrectangles=True)
```

[image]

I don't think there is anything to fix at our end, perhaps it's worth filing an upstream bug report with `imageio`. But this fix seems simple enough. @alexrockhill can you make a PR to PySurfer to amend the `save_movie` docs to mention this can be useful for `gif` format, as can looking at `imageio.help('gif')` (or whatever the output-specific options are for your chosen output)?

closed 2018-10-17 by larsoner

alexrockhill, 2018-10-17:
Sure, will do the PR. Thanks for the work-around!

alexrockhill, 2018-10-18:
Using subrectangles = True, gives me some really weird results. See the home page of https://github.com/alexrockhill/MEEGbuddy for instance. Any idea what's going on?

larsoner, 2018-10-18:
Some conversion problem. I'd try saving to another format.

alexrockhill, 2018-10-22:
For my application, I just wrote custom code to make gif with the colorbar stationary. For reference, the code is below as well as in aforementioned MEEGbuddy github page. If you want to incorporate this into stc.plot, I would be willing to submit a pull request but otherwise, I'm assuming most other people will not need a gif, and so the documentation could just say save in another format if using the colorbar. 

import numpy as np
import matplotlib.pyplot as plt
from matplotlib import animation, rc
from PIL import Image

fig,ax = plt.subplots()
gif = Image.open('test.gif')
width,height = gif.size
colorbar_box = (int(0.0625*width),int(0.9*height),
                    int(0.9275*width),height)
im = ax.imshow(gif.copy().convert('RGB'))

def setColorbar(im_data):
        gif.seek(0)
        im_data2 = gif.copy().convert('RGB')
        for i in range(colorbar_box[0],colorbar_box[2]):
            for j in range(colorbar_box[1],colorbar_box[3]):
                value = im_data2.getpixel((i,j))
                value2 = im_data.putpixel((i,j-height_offset),value)
        return im_data

def animate(i)::
    gif.seek(i)
    im_data = setColorbar(im_data)
    im.set_data(im_data)
    return im

anim = animation.FuncAnimation(fig,animate,frames=frames,interval=10,blit=True)
anim.save(output_fname,fps=fps,writer='imagemagick',
                 savefig_kwargs={'facecolor':'black'})