mne-tools/mne-python #5607: [Feature Request] expose original units for each channel in BaseRaw

sappelhoff, 2018-10-13:
Upon reading a raw data file, MNE-Python converts the units to e.g., Volts (and others, see [here](http://martinos.org/mne/stable/manual/io.html)).

It would be good to expose the channel wise units as they are in the original data file within the `BaseRaw` object.

Concretely, we need this in MNE-BIDS, see: https://github.com/mne-tools/mne-bids/issues/81

agramfort, 2018-10-13:
you need this for what file format?

sappelhoff, 2018-10-14:
for EEG:
- BrainVision (.vhdr / .vmrk / .eeg)
- EDF / BDF 
- Neuroscan (.cnt)
- EEGLAB (.set / .fdt)

for MEG it depends a bit in BIDS, all MEG file formats are accepted (i.e., no restrictions like we have in EEG). So for MEG, we need the requested feature for all file formats, that natively save data in a unit that is different from the one that MNE-Python converts to.

agramfort, 2018-10-14:
you need to have this availble in the raw object or just a public function
like read_xxx_header than you can use in mne-bids?

sappelhoff, 2018-10-14:
both would be fine, but a `read_xxx_header` public function could potentially return a lot more information and could be helpful for other issues as well.

On the other hand, this would mean many new functions (one per data format), while making the information available in `BaseRaw` would need adding a few private lines of code within each data reader in the IO module.

Are you more inclined to having a `read_xxx_header` function for each data format?

larsoner, 2018-10-14:
`read_xxx_header` would probably require refactoring raw readers and, probably more annoyingly, introducing a bunch of otherwise unnecessary public functions. After all, `read_xxx_raw(..., preload=False)` is already very close to what the `read_xxx_header` would do and `raw.info` is very close to what it would return (though also annoyingly, not quite identical because it would also need to give the original channel types). So I don't see the point of a separate function, it seems cleaner to expose a `raw._orig_units` or something that is used only by BIDS.

agramfort, 2018-10-14:
ok then. Why not for _orig_units attributes. Note that there is also the
cals array that scales the data to V so maybe this can he used too.

sappelhoff, 2018-10-19:
Listing the lines of the respective data readers, from where the original units could be forwarded to some `_orig_units` attribute.

# BrainVision

# EDF
https://github.com/mne-tools/mne-python/blob/f4096adaf2646da32e6d2f5536e74c14b312ad5f/mne/io/edf/edf.py#L639

# Neuroscan

# EEGLAB

agramfort, 2018-10-19:
can you give a try to a PR?