mne-tools/mne-python #5611: ENH: ability to pass a list of files to `read_raw_*`

wmvanvliet, 2018-10-16:
#### Describe the problem
During long recording sessions, raw files are often split up into parts. In the FIFF format, these parts will link to each other and MNE-Python will read all of them. However, maxfilter destroys this link and other fileformats don't have such functionality in the first place.

#### Describe your solution
The various `read_raw_*` functions should also be able to take a list of filenames to read. The raw files are appended together (first with `preload=False`, then if preloading was desired, a single call to `load_data()` on the joined `Raw` object.)

The required code for this is already mostly present in mne/io/raw/fiff.py#83

larsoner, 2018-10-16:
-1 on this, we already have `concatenate_raws` and `raw.append` to take care of this. The fact that we can do or get close to this list-like behavior in the FIF code is either a historical accident or something special about the file format (can't remember which) -- either way, not something we should advertise / advocate over the standard existing method.

agramfort, 2018-10-16:
I agree with @larsoner

we should try to have only one recommended way to do things.

closed 2018-10-16 by wmvanvliet