mne-tools/mne-python #5614: Epochs filtering error likely due to save/load data typing

alexrockhill, 2018-10-16:
raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif')
events = mne.find_events(raw)
epochs = mne.Epochs(raw,events)
epochs.save('test-epo.fif')
epochs = mne.read_epochs('test-epo.fif')
epochs = epochs.filter(l_freq=None,h_freq=40)

Gives the following error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 2, in filter
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/utils.py", line 803, in verbose
    return function(*args, **kwargs)
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/filter.py", line 2045, in filter
    fir_window=fir_window, fir_design=fir_design, pad=pad)
  File "<string>", line 2, in filter_data
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/utils.py", line 803, in verbose
    return function(*args, **kwargs)
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/filter.py", line 859, in filter_data
    copy, pad)
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/filter.py", line 164, in _overlap_add_filter
    x, orig_shape, picks = _prep_for_filtering(x, copy, picks)
  File "/autofs/space/karima_001/users/alex/software/anaconda3/lib/python3.6/site-packages/mne-0.17.dev0-py3.6.egg/mne/filter.py", line 269, in _prep_for_filtering
    raise TypeError("Arrays passed for filtering must have a dtype of "
TypeError: Arrays passed for filtering must have a dtype of np.float64

When I look at the dtype of the epochs before being saved it's float64 and after it's dtype('>f8') so I would hazard to guess that this is the issue.

agramfort, 2018-10-16:
I confirm the pb with current master :(

@larsoner can you have a look? it's related to our recent support for
complex numbers in epochs fif files. We should force conversion to
double precision when putting in memory.

larsoner, 2018-10-17:
That is double precision it's just a different endianness. Should be an easy fix, yes I can look

PR #5625 (2018-10-18): MRG: Fix filtering check
Closes #5614.

@alexrockhill can you verify this fixes your issue?

I also replaced `_TempDir` calls with pytest `tmpdir` fixture, something that I think we can do gradually as we find them.
closed 2018-10-18 by agramfort