mne-tools/mne-python #5615: Add Read .bvct montage files

alexrockhill, 2018-10-16:
I added the functionality to the read_dig_montage file below, but I was not able to write tests because there are no MNE sample .bvct files. I attached a sample .bvct file and a test could be the following if a bvct file could be made from the raw montage. It would be quite a bit of work however to reverse engineer a bvct file.

----------------------------
test
----------------------------
import mne
raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif')
montage = mne.channels.read_dig_montage(bvct=sample)
raw = raw.set_montage(montage,verbose=False)

-------------------------------------------------------------------
functionality added
-------------------------------------------------------------------

def read_dig_montage(hsp=None, hpi=None, elp=None, point_names=None,
                     unit='auto', fif=None, egi=None, bvct=None, transform=True,
                     dev_head_t=False):
    r"""Read subject-specific digitization montage from a file.

    Parameters
    ----------
    hsp : None | str | array, shape (n_points, 3)
        If str, this corresponds to the filename of the headshape points.
        This is typically used with the Polhemus FastSCAN system.
        If numpy.array, this corresponds to an array of positions of the
        headshape points in 3d. These points are assumed to be in the native
        digitizer space and will be rescaled according to the unit parameter.
    hpi : None | str | array, shape (n_hpi, 3)
        If str, this corresponds to the filename of Head Position Indicator
        (HPI) points. If numpy.array, this corresponds to an array
        of HPI points. These points are in device space, and are only
        necessary if computation of a ``dev_head_t`` by the
        :class:`DigMontage` is required.
    elp : None | str | array, shape (n_fids + n_hpi, 3)
        If str, this corresponds to the filename of electrode position
        points. This is typically used with the Polhemus FastSCAN system.
        If numpy.array, this corresponds to an array of digitizer points in
        the same order. These points are assumed to be in the native digitizer
        space and will be rescaled according to the unit parameter.
    point_names : None | list
        A list of point names for elp (required if elp is defined).
        Typically this would be like::

            ('nasion', 'lpa', 'rpa', 'CHPI001', 'CHPI002', 'CHPI003')

    unit : 'auto' | 'm' | 'cm' | 'mm'
        Unit of the digitizer files (hsp and elp). If not 'm', coordinates will
        be rescaled to 'm'. Default is 'auto', which assumes 'm' for \*.hsp and
        \*.elp files and 'mm' for \*.txt files, corresponding to the known
        Polhemus export formats.
    fif : str | None
        FIF file from which to read digitization locations.
        If str (filename), all other arguments are ignored.

        .. versionadded:: 0.12

    egi : str | None
        EGI MFF XML coordinates file from which to read digitization locations.
        If str (filename), all other arguments are ignored.

        .. versionadded:: 0.14

    bvct : srt | None
        BVCT XML coordinates file from which to read digitization locations.
        (BrainVision)
        If str (filename), all other arguments are ignored.

    transform : bool
        If True (default), points will be transformed to Neuromag space
        using :meth:`DigMontage.transform_to_head`.
[... 1964 lines cut ...]
      <Name>Nasion</Name>
      <Number>0</Number>
      <Group>0</Group>
      <Theta>112.152588</Theta>
      <Phi>90</Phi>
      <Radius>0.9998025</Radius>
    </ActiCapElectrode>
    <ActiCapElectrode>
      <Name>LPA</Name>
      <Number>0</Number>
      <Group>0</Group>
      <Theta>-112.152588</Theta>
      <Phi>0</Phi>
      <Radius>0.9998025</Radius>
    </ActiCapElectrode>
    <ActiCapElectrode>
      <Name>RPA</Name>
      <Number>0</Number>
      <Group>0</Group>
      <Theta>112.152588</Theta>
      <Phi>0</Phi>
      <Radius>0.9998025</Radius>
    </ActiCapElectrode>
  </ElectrodeLayout>
</BrainVisionCapTrakFileV1>

agramfort, 2018-10-16:
please open a pull request and see the guidelines we offer for contributing.

thanks

alexrockhill, 2018-10-16:
Hi Alex,

   I just submitted a pull request, I just added more detail about
potentially adding a test in an issue.

Thanks,

Alex

agramfort, 2018-10-17:
can you share a tiny file so i can have a look?

alexrockhill, 2018-10-17:
Sure, here is a gist https://gist.github.com/alexrockhill/d6dacee5efe20bac76ab81f0438f5667. Sorry I think the copy and paste in the issue text ruined the formatting.

alexrockhill, 2018-10-17:
Just to be clear, this is a montage file but it's for a 96 channel EEG setup so it won't work with the sample MNE raw files as their montage. There lies the rub of producing a test, either I need a bvct file from an existing montage file in another format for a MNE raw sample file or (not an option, I know) I would need to add a raw sample file for which I have the bvct montage. Maybe you can think of a way around this.

agramfort, 2018-10-19:
I would trim down this file to keep only LPA, RPA, NASION, GRD and FPz in
the xml file
then use this much shorter file for testing.

see this
https://github.com/mne-tools/mne-python/blob/master/mne/channels/tests/test_montage.py#L136
for inspiration

larsoner, 2020-07-26:
This was in #6663 I think

closed 2020-07-26 by larsoner