mne-tools/mne-python #5618: Modify subprocess error/info

ktavabi, 2018-10-16:
On a machine without `tcsh` installed `run_subprocess(['mne', 'make_scalp_surfaces', '--subject', subject])` returns following:

> Command not found: mkheadsurf
...
subprocess.CalledProcessError: Command '['mne', 'make_scalp_surfaces', '--overwrite', '--subject', 'sample', '--force']' returned non-zero exit status 1.
> ('', '')

Would a PR to make error message more meaningful be useful?

agramfort, 2018-10-17:
I don't think it's related to tcsh. Is freesurfer sourced properly in
the tcsh shell?

ktavabi, 2018-10-17:
I believe so. Have always sourced it as `source $FREESURFER_HOME/SetUpFreeSurfer.sh`. But now we can't reproduce it on the said machine after removing `tcsh`.  Must have been a ghost between keyboard and chair 😕 Sorry for the inconvenience. Closing this down.

ktavabi, 2018-10-18:
@agramfort FWIW I uninstalled `tcsh` and encountered this TB again, and resolved once tcsh was reinstalled. Not sure what this can be?

> Traceback (most recent call last):
>   File "/home/nordme/GitHub/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 133, in <module>
>     run()
>   File "/home/nordme/GitHub/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 60, in run
>     options.no_decimate, options.verbose)
>   File "<string>", line 2, in _run
>   File "/home/nordme/GitHub/mne-python/mne/utils.py", line 797, in verbose
>     return function(*args, **kwargs)
>   File "/home/nordme/GitHub/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 94, in _run
> Command not found: mkheadsurf
>     env=this_env)
>   File "<string>", line 2, in run_subprocess
>   File "/home/nordme/GitHub/mne-python/mne/utils.py", line 797, in verbose
>     return function(*args, **kwargs)
>   File "/home/nordme/GitHub/mne-python/mne/utils.py", line 1304, in run_subprocess
>     p = subprocess.Popen(command, *args, **kwargs)
>   File "/home/nordme/miniconda3/envs/mne/lib/python3.6/subprocess.py", line 709, in __init__
>     restore_signals, start_new_session)
>   File "/home/nordme/miniconda3/envs/mne/lib/python3.6/subprocess.py", line 1344, in _execute_child
>     raise child_exception_type(errno_num, err_msg, err_filename)
> FileNotFoundError: [Errno 2] No such file or directory: 'mkheadsurf': 'mkheadsurf'
> Traceback (most recent call last):
>   File "/home/nordme/GitHub/Scripts/run_fs_recon_mne_bem.py", line 142, in <module>
>     '--overwrite', '--subject', subject, '--force'])
>   File "<string>", line 2, in run_subprocess
>   File "/home/nordme/GitHub/mne-python/mne/utils.py", line 797, in verbose
>     return function(*args, **kwargs)
>   File "/home/nordme/GitHub/mne-python/mne/utils.py", line 1321, in run_subprocess
>     raise subprocess.CalledProcessError(p.returncode, command, output)
> subprocess.CalledProcessError: Command '['mne', 'make_scalp_surfaces', '--overwrite', '--subject', 'genz102_9a', '--force']' returned non-zero exit status 1.
> ('', '')
> Process finished with exit code 1

agramfort, 2018-10-19:
if you call mne shell scripts or freesurfer they were written for tcsh