mne-tools/mne-python #5623: open new issue from the code no longer works

massich, 2018-10-17:
Normally issues can be open from the code. 

i.e: get a line in mne-tools/mne-openmeeg, clik and open an issue:

[image]
[image]

However in mne-tools/mne-python,
[image]
you get to the issue template selection:
[image]
and even if you try `Open a regular issue` the issue is empty
[image]

Does anyone know how to fix/trace that? 
Also if someone thinks this irrelevant, feel free to close this issue.

sappelhoff, 2018-10-18:
I have encountered this as well and it's been there ever since we introduced the templates. Would be nice to fix it ... +1

larsoner, 2018-11-07:
Closing this as an upstream bug. I think we have to live with it until they fix it unless we want to scrap the templates

closed 2018-11-07 by larsoner