mne-tools/mne-python #5627: no "overwrite" parameter in the epoch.save method

QB3, 2018-10-17:
#### Describe the problem
* The `save` method of the `raw` object has  parameters   `fname` (string) and `overwrite` (boolean) stating if you want to overwrite a file with the name `fname`.
* However the `save` method from the `epoch` object has no such `overwrite` parameter.
* I would expect the API to behave the same. 

See these two parts of `plot_introduction.py`:

https://github.com/mne-tools/mne-python/blob/9df12aa57dce2a75ba619c90abc6a6547dde9997/tutorials/plot_introduction.py#L156-L157

https://github.com/mne-tools/mne-python/blob/9df12aa57dce2a75ba619c90abc6a6547dde9997/tutorials/plot_introduction.py#L243

and this is the `Epochs.save(...` signature :
https://github.com/mne-tools/mne-python/blob/9df12aa57dce2a75ba619c90abc6a6547dde9997/mne/epochs.py#L1680

#### Describe your solution
It would be good if a  boolean parameter `overwrite` could be added to the method `save` of the `epoch` object to state if you want to overwrite a file or not.

larsoner, 2018-10-17:
@QB3 can you make a PR to add this option?

agramfort, 2018-10-19:
PR very welcome indeed !

massich, 2018-10-23:
@QB3 https://mne-tools.github.io/stable/contributing.html are the instructions in how to contribute, if something is not clear enough. Please open an issue and we'll try to improve our documentation. Thanks a lot.

PR #6108 (2019-04-05): ENH: Added overwrite parameter to epochs.save
#### Reference issue
Fixes #5627

#### What does this implement/fix?
I added an `overwrite` parameter to `epochs.save`, similar to that in `raw.save`.

#### Additional information
 I set the default to be `overwrite=False`, for consistency with `raw.save`. This changes the behavior of `epochs.save`. Is this correct or should I add a deprecation warning?

I had to add `overwrite=True` as an argument to `epochs.save` in a number of unit tests, so that the behavior would be consistent with previous versions of `epochs.save`. Please also let me know if [... cut]
closed 2019-04-05 by jona-sassenhagen