mne-tools/mne-python #5629: Line not reached in BrainVision

sappelhoff, 2018-10-18:
I don't have a concrete bug report, but while scanning the BrainVision reader for a good way to solve #5607, I found these lines:

https://github.com/mne-tools/mne-python/blob/f4096adaf2646da32e6d2f5536e74c14b312ad5f/mne/io/brainvision/brainvision.py#L690-L697

Note that line 696 and 697 are never reached due to lines 690 and 691.

larsoner, 2018-10-19:
I'm not convinced these can never be reached, `props = (..., ..., ..., '')` would reach these lines

sappelhoff, 2018-10-19:
okay, thanks @larsoner I wasn't aware that an empty string evaluates to None. Sorry for pulling your attention needlessly Teon

closed 2018-10-19 by sappelhoff

larsoner, 2018-10-19:
Evaluates to `False` in a boolean, yeah (in general any built-in with `len(...) == 0` will do so).