mne-tools/mne-python #5630: easy to find mne-c commands download button.

massich, 2018-10-19:
The download button of MNE-C commands is kind of hard to find. Maybe we should place a big colorful button somewhere. 

https://github.com/mne-tools/mne-python/blob/master/doc/install_mne_c.rst#L8

drammock, 2019-09-19:
MNE-C is still behind a login wall.  We can add a bigger button I guess, but it will still go [here](http://www.nmr.mgh.harvard.edu/martinos/userInfo/data/MNE_register/index.php).  @massich given how the MNE-C install page looks now, is this still something you want implemented?

[image]

larsoner, 2019-09-21:
It looks clear enough given the new title/org so I'll close, but feel free to reopen if you disagree @massich

closed 2019-09-21 by larsoner