mne-tools/mne-python #5631: [DOC] forward.rst pointing to C API

massich, 2018-10-19:
When rendered, the [this](https://github.com/mne-tools/mne-python/blob/master/doc/manual/source_localization/forward.rst#creating-a-surface-based-source-space) snipped (copied at the end and that renders as [this](https://mne-tools.github.io/dev/manual/source_localization/forward.html#creating-a-surface-based-source-space)) ``:ref:`mne_make_source_space` `` points to the C API here: https://mne-tools.github.io/dev/manual/c_reference.html#mne-make-source-space

```
Creating a surface-based source space
#####################################

The fif format source space files containing the dipole locations
and orientations are created with the utility :ref:`mne_make_source_space`.
This utility is usually invoked by the convenience script :ref:`mne_setup_source_space`,
see :ref:`setting_up_source_space`.
```

This happens in different places.

hubertjb, 2018-10-19:
Just checked the link, and here are all the C API references:

- mne_make_source_space
- mne_setup_source_space
- mne_surf2bem
- mne_setup_forward_model
- mne_prepare_bem_model
- mne_do_forward_solution
- mne_forward_solution
- mne_average_forward_solutions

larsoner, 2018-10-19:
They should probably contain refs to the MNE-Python version and the MNE-C version.

agramfort, 2018-10-19:
yes this file needs to be rewritten to point first to the python version

massich, 2018-10-19:
> yes this file needs to be rewritten to point first to the python version

yes, its hard to follow and the instructions are not propper. (i.e: it tells you to run a script and to set the variables, that this script needs afterwards.)

msrdinesh, 2019-02-25:
Hello, is the issue solved? I think forward.rst file has been changed.  I could not find the error in [this](https://github.com/mne-tools/mne-python/blob/master/doc/manual/source_localization/forward.rst#creating-a-surface-based-source-space) link which was mentioned. If any work needs to be done on this issue regards, please assign me the work. I will try to do it!

massich, 2019-03-20:
As far as I can tell, this was solved.

closed 2019-03-20 by massich