mne-tools/mne-python #5635: Install MNE-C using ZSH

massich, 2018-10-19:
I'm trying to install **MNE-C** (`MNE-2.7.3-3268-Linux-x86_64`) in this machine:
```sh
~/software
(base) ❯ uname -a                   
Linux toothless 4.15.0-34-generic #37-Ubuntu SMP Mon Aug 27 15:21:48 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
using a `ZSH` shell.

It is known it is known that `zsh`, `bash`, `tcsh` and other shell languages differs but `#!/bin/blabla` in the top of the scripts should account for that. But I don't really think is working in my system. 

the documentation says to execute:
```sh
$ . $MNE_ROOT/bin/mne_setup_sh
$ . $MNE_ROOT/bin/mne_setup
```
but this wont work. However, this should

```sh
$ bash $MNE_ROOT/bin/mne_setup_sh
$ tcsh $MNE_ROOT/bin/mne_setup
```

Can anyone try to replicate. It was not working, I tried lots of other things now it seems to work and I can't replciate the problems. 

### Things that might be useful
- when switching to bash make sure that `env | grep -i shell` outputs `/bin/bash` and not `/bin/zsh`. If it doesn't do `export SHELL=/bin/bash` before doing anything else

- if you see an error like this:

```sh

sik@toothless:~/software$ . $MNE_ROOT/bin/mne_setup
bash: /home/sik/software/MNE-2.7.3-3268-Linux-x86_64/bin/mne_setup: line 75: syntax error near unexpected token `else'
bash: /home/sik/software/MNE-2.7.3-3268-Linux-x86_64/bin/mne_setup: line 75: `else if( "x$OS" == "xDarwin" ) then'
```
change this

```diff
-    if(! $?DYLD_LIBRARY_PATH ) then
+    if ( ! $?DYLD_LIBRARY_PATH ) then
```

---
I'll try to re do it from scratch but right now I've no idea of the state of my machine 'cos it seems to work, before it wasn't and I don't understand the difference.

agramfort, 2018-10-20:
cc @buildqa @mshamalainen

buildqa, 2018-10-22:
The hash bang at the top of a script  tells exec what program to use by default to interpret the script.  But the hash bang line is ignored when you source a script via either the builtin "source" or "." directive.  (BTW "." will search for the script in $PATH if there is no slash in the script argument while "source" I don't think does a $PATH search).  So while in csh you can run a bash script, e.g., foo.sh, in csh as "bash foo.sh" (hash bang overriden) or "./foo.sh" (should use the hash bang), there is no way to source bash script foo.sh in csh. 

 Zsh is supposed to be bash compatible and run bash scripts, but on my CentOS7 machine, the mne_setup_sh script exits with an error if I try to run it with bash.   It does however run if I invoke it with "sh".   zsh has an emulate builtin you can use to run sh and csh scripts.   It worked for me in zsh to source the mne_setup_sh with the command,

% emulate sh -c  'source $MNE_ROOT/bin/mne_setup_sh'

massich, 2018-10-26:
Some custom functions just to make sure that I've MNE-python, matlab and freesurfer. (happy to share them if needed)
```sh
~/software/
(base) ❯ conda activate mne && activate_matlab && freesurfer
-------- freesurfer-Linux-centos6_x86_64-stable-pub-v6.0.0-2beb96c --------
Setting up environment for FreeSurfer/FS-FAST (and FSL)
FREESURFER_HOME   /home/sik/software/freesurfer
FSFAST_HOME       /home/sik/software/freesurfer/fsfast
FSF_OUTPUT_FORMAT nii.gz
SUBJECTS_DIR      /home/sik/software/freesurfer/subjects
MNI_DIR           /home/sik/software/freesurfer/mni
```

Here is what I did to install MNE-C
```sh
~/software
(mne) ❯ tar zxf MNE-2.7.3-3268-Linux-x86_64.tar.gz 

~/software
(mne) ❯ export MNE_ROOT=$PWD/MNE-2.7.3-3268-Linux-x86_64

~/software
(mne) ❯ emulate sh -c 'source $MNE_ROOT/bin/mne_setup_sh'

MNE software location set to:    /home/sik/software/MNE-2.7.3-3268-Linux-x86_64
MATLAB software location set to: /opt/matlab/R2017b

/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/bin already in PATH
/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/lib added to LD_LIBRARY_PATH
/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/share/app-defaults/%N added to XUSERFILESEARCHPATH

Note: Remember to set SUBJECTS_DIR and SUBJECT environment variables correctly.
Note: FreeSurfer environment is needed to access tkmedit from mne_analyze.

~/software
(mne) ❯ tcsh $MNE_ROOT/bin/mne_setup   

MNE software location set to:    /home/sik/software/MNE-2.7.3-3268-Linux-x86_64
MATLAB software not available

/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/bin already in PATH
/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/lib added to LD_LIBRARY_PATH
/home/sik/software/MNE-2.7.3-3268-Linux-x86_64/share/app-defaults/%N added to XUSERFILESEARCHPATH

Note: Remember to set SUBJECTS_DIR and SUBJECT environment variables correctly.
Note: FreeSurfer environment is needed to access tkmedit from mne_analyze.

~/software
(mne) ❯ mne_analyze                                                  
mne_analyze: error while loading shared libraries: libXp.so.6: cannot open shared object file: No such file or directory
```

and we end up in the `libXp.so.6` error which is known. (see #5650).

Can anyone check that those steps work for them (especially people using a system different than mine)?
@buildqa can you open a PR updating the documentation?

buildqa, 2018-10-29:
You could try running the ldd command on the mne_analyze binary to see the default search path to libXp.so. , and check if it is there.  I'll take a look at the documentation.

massich, 2018-11-07:
close via #5687

closed 2018-11-07 by massich