mne-tools/mne-python #5637: inverse operator’s source space in 3D

erfpak7, 2018-10-19:
I would like to compute sLoreta and Inverse solution by which I can make a 3D source space as follow:
http://www.martinos.org/mne/stable/auto_examples/inverse/plot_read_inverse.html

However, I do not have a .fif file and I just use a .txt file converted to MNE structure. It would be great if someone lets me know how I can make an inverse operator output without fif file. Because even the **"apply_inverse_raw"** function needs inverse operator.
Thanks in advance

agramfort, 2018-10-20:
you want to do source localization without MRI using only EEG standard locations?

erfpak7, 2018-10-21:
> you want to do source localization without MRI using only EEG standard locations?

@agramfort Yes, that's what I'm looking for

agramfort, 2018-10-22:
then it's not straightforward to do with MNE currently. Sorry.

Maybe someone else can share some tips to do it but there is still no
tutorial to achieve what you want.

mmagnuski, 2018-10-24:
Currently you would still need to go through all the steps of coregistration and building forward model. In my case that took some time, especially at the beginning: constructing 'fake' electrode digitization (I found that the docs for digitization class could be clearer on some points), then digitizing the channel position with photogrammetry (in one study it was just for one representative head - because I needed fiducial position relative to the channels), then figuring out some stuff in coreg gui - and after that it went downhill. Unfortunatelly I didn't write down a narrative of the whole process so I can't offer much at the present moment except a word of encouragement and a promise that it should get easier in the future.

larsoner, 2023-10-01:
This is now possible without too much effort, see https://mne.tools/stable/auto_tutorials/forward/35_eeg_no_mri.html

closed 2023-10-01 by larsoner