mne-tools/mne-python #5640: Add option to save annotations as plain text

cbrnr, 2018-10-22:
Currently, annotations can only be saved to a FIFF file. Since this is a format that can't be read outside of MNE easily, an option to save annotations as plain text would be very useful. I can work on this if you think this should be added (e.g. as an optional argument to `annotations.save`, for example `format="CSV"` or `kind=`text` or `type=`TXT` - let me know what you prefer). @massich what do you think?

agramfort, 2018-10-22:
yes +100 to have a way to write them as CSV.

question is how to store the date/instant/onset wrt orig_date ? should we
do it the brainvision way?
we should be able to write time stamps I think.

thoughts?

cbrnr, 2018-10-23:
I never needed to set this value because annotations are always tied to a specific raw file for me. What is the use case of writing out a specific value instead of always taking onset 0 (in the annotations object) to mean the start of the raw recording? For example, if I do `raw.crop`, annotation onsets are automatically adjusted, so there should be no need for this field.

cbrnr, 2018-10-23:
> if you store annots and raw in different files and the raw get cropped or resampled you're screwed.

You mean if you first store the annotations in a separate file, and then you manipulate the raw file? Even when storing `orig_time` with the annotations, there is no way resampling can be accounted for. That's why I'd just export the annotations after these processing steps, because if the annotations are stored in `raw.annotations`, they get updated automatically.

> Also annots eg for sleep are during night and have the timestamp is important

The time is stored in the raw object. I would think that a processing pipeline will always associate annotations with a raw object, which already has all the information. I don't see why we would have to duplicate the information and store the time with the annotations.

So the use case you're mentioning is that you have annotations made on some raw object. You store these annotations in a separate file. Then you go ahead and modify the raw object (e.g. cropping, resampling), at which point the annotations do not match the new raw object anymore.

This is only a problem if you then only have the cropped/modified raw object and the original annotations. This shouldn't ever be a thing, because you can export annotations after every processing step.

Anyway, maybe I don't get the use case, but that's what storing annotations in FIFF files should already take care of. What do you think of having a "lightweight" export option, meaning that if you export annotations as CSV only onset, duration, and descriptions are exported (a warning message could make this obvious to the user)?

jasmainak, 2018-10-23:
> Even when storing orig_time with the annotations, there is no way resampling can be accounted for.

Just thought I'd pitch in to the discussion here. You do have `meas_date` stored in the `raw.info`, so if you store the absolute time in the annotation file, you could in principle sync them again. This is related to discussion we had with @agramfort for BIDS derivatives. Although thinking about this again, this works only if these annotations are stored with time stamps up to millisecond/microsecond precision.

larsoner, 2018-10-23:
> What do you think of having a "lightweight" export option, meaning that if you export annotations as CSV only onset, duration, and descriptions are exported (a warning message could make this obvious to the user)?

I don't see why we wouldn't also include `meas_date` in there in a comment, like:
```
# MNE Annotations CSV
# orig_time: (X, Y)
# onset, duration, description
502.45682716, 0, BAD_EDGE
...
```
Then on read if it's there, use it, if it's not, it's the same as constructing Annotations with `orig_time=None`

agramfort, 2018-10-23:
I would like these CSVs as decoupled from MNE objects as possible so they
can be shared, reused and imported from other software. Also the CSV should
be standard so I can do

import pandas as pd
df = pd.read_csv('annotations.csv')
df['onset'] = pd.to_datetime(df['onset'])

and get proper time stamps.

larsoner, 2018-10-23:
Most readers (pandas included) have an option to skip rows and/or get names from a specific row, so you would just need a few `kwargs` there. But if you don't want to have the two MNE lines at the top I can live with it and just have:
```
# onset, duration, description
502.45682716, 0, BAD_EDGE
...
```

agramfort, 2018-10-23:
I don't think the # is necessary on the first line and here we lost
orig_time unless you have a way to "interpret" 502.45682716 as
timestamp?

larsoner, 2018-10-23:
Yes we have lost `orig_time` but I don't see a way to do all 3 of these:

1. Have `orig_time` represented
2. Have no extra comment lines at the top
3. Don't risk losing precision
4. Easily usable by the end user

I think we have to choose 3 if we want to use CSV. My proposal with the `# orig_time: ` line chooses (1), (3) at the expense of (2) and perhaps some of (4).

If you're proposing including the offset since 1970 in the timestamps (?), then we have (1) and (2), but at the risk of (3) losing precision at some point, depending on the time units and time since 1970 (or something like this). And if you include this offset the data are now much less readable, so it also hurts (4).

The only way I think we can have 1-3 is to make it more complicated, namely by having four columns:
```
onset_sec, onset_microsec, duration, description,
```
where `onset = onset_sec + onset_microsec * 1e-6`, i.e., the same format as our `meas_date` usually works. But this comes at the expense of 4.

jasmainak, 2018-10-23:
@larsoner in your proposal, is `onset_microsec` also in the "date-time" format? could that be a float? 

I'm thinking that eventually this kind of file should be compatible with BIDS derivative. So, perhaps it should be in TSV rather than CSV (just a different delimiter)?

larsoner, 2018-10-23:
> onset_microsec also in the "date-time" format? could that be a float?

No it's like `meas_date` where both are ints

massich, 2018-10-23:
If I would have to pick a way of doing it I would do it as @larsoner proposed. A header with `meas_date` and then onsets. Basically, a dump of our Annotations object. 

Then we can discuss if the writer should just dump the Annotation or reset to `first_samp` and write the `onsets` relative to this time point. If when we do this resampling to `first_samp` would have the option of dropping the header all together as a manner of saying that the onsets are relative to `first_samp`. Whatever. 

But the base case. I strongly advocate for header + data. (In the header I would also state format version, and number of heather lines. Although it could be argued if they are necessary or not.)

agramfort, 2018-10-23:
how about this :

https://gist.github.com/agramfort/320c031735e7aee1b3f46c35eb3426e9

?

cbrnr, 2018-10-24:
I'd still prefer a simpler CSV variant where 0 is the start of the recording. Maybe we can make this optional.

Regarding @agramfort's suggestion, this looks good, but we should still include the time stamp corresponding to time 0 in the header. That way, people could subtract that time from all onsets when the following header line is present:

    # orig_time: 2002-12-03 19:01:11.720100

Comment lines can be easily ignored by pandas and readr R packages (`comment="#"`), but of course the user will have to parse this line manually then. So in the end, I'd still prefer an option to export onsets with `orig_time` already subtracted, so that 0 means start of the recording (again, this could be optional behavior).

agramfort, 2018-10-24:
adding a header in a csv file seems really a hack. It will be not be understood by default csv readers or excel or google sheets.

do you have example where such a header is employed?

@larsoner I am not sure why 4 will not be satisfied. Read the csv with pandas and then raw.set_annotations(annotations) should do a perfect IO round trip.

cbrnr, 2018-10-24:
It opens fine with Excel (I just double-click), all columns are correctly recognized and the comment is just a comment line.

larsoner, 2018-10-24:
I comment about 4 because it's easy in Pandas and Excel, by harder in simpler programs that do not automatically parse a date string. Those will have to do string and date gymnastics to get a usable time. Think about what it would take in Python with no  Pandas -- it's only a few lines but certainly more work than a sample number or time offset in seconds relative to zero would be.

agramfort, 2018-10-24:
It seems it may not be so painful in matlab either:

https://fr.mathworks.com/matlabcentral/answers/362363-how-do-i-read-this-csv-file-in-and-convert-the-dates

I my concern is that annotations should be as easy to exchange as possible.
No matter how have been cropped or resampled or processed raw data.

larsoner, 2018-10-24:
Yes but it adds more work. It's not a deal breaker, though, and I agree so far a `str` date stamp is the cleanest / least hacky solution even though it doesn't perfectly satisfy (4).

One other thing we could do is *just for text I/O* (because we don't need it for FIF as there is a separate field) add as the first entry:
```
<whenever>,0,ORIG_TIME
```
```
I *think* this would make the information in `-annot.fif` and `.txt` outputs equivalent.

And somewhere -- probably in the Annotations docstring -- we start making a list of all special/reserved annotation names, such as `BAD_*` and `EDGE boundary` and this new `MEAS_DATE`.

agramfort, 2018-10-24:
I think we need a use case to see what works best and I agree that we can
have different formats .fif .csv .txt (maybe .vmrk)
to satisfy everyone

larsoner, 2018-10-24:
@cbrnr can you try what is in @agramfort's gist and see if it's usable enough for someone to turn into a PR?

You can add the `ORIG_TIME` line, too, if you want (so you can do time deltas, etc. easily)

massich, 2018-10-24:
I'll open a PR tomorrow (unless someone else whats to go for it)

cbrnr, 2018-10-25:
Like I said, and here I agree with @larsoner, I would really love to have a simple file where the onset column contains time (or sample) offsets relative to zero. This will make it much easier to work with, otherwise I have to to all kinds of date conversions - which might even introduce additional variance in the time points. I think it's a good idea to have several options because apparently there are several use cases out there.

jona-sassenhagen, 2018-11-07:
`annotation.to_data_frame().to_csv(name)`?

massich, 2018-11-07:
So far, annotations are saved only as FIF. I would start by adding the reader #5699

jona-sassenhagen, 2019-03-24:
Closed for #5699

closed 2019-03-24 by jona-sassenhagen