mne-tools/mne-python #5642: BUG: Manipulating the times property of Epochs results in wrong data after average if data was not loaded.

thht, 2018-10-22:
#### Describe the bug
An `Epochs` object has a `times` property that can be modified. The effect of modifying this property is inconsistent, depending on whether the `Epochs` object has preloaded data (i.e. `load_data`) was called or not.

I created a test for this bug: https://github.com/thht/mne-python/blob/th/test_timeshifting/mne/tests/test_timeshift.py

agramfort, 2018-10-22:
you should not be allowed to do: epochs1.times += t_shift

epochs.times should be read only as it's obtained from tmin, tmax and sfreq

can you open a PR to make sure epochs.times is read only?

thht, 2018-10-23:
ok, i can do that. however, i have a current use case in which it would be very beneficial for the logical flow of my analysis to be able to correct for a delay introduced during stimulation (auditory stimulation with tubes). this can be done for instances of `Evoked` but in my case i do single-trial analysis.

so, do you think there is a way to make times read only but introduce a method like `Evoked.shift_time` for `Epochs`?

agramfort, 2018-10-23:
no objection to add shift_time method in Epochs. Would this be enough for
you use case?

larsoner, 2018-10-23:
Take a look at what `epochs.decimate` and `epochs.resample` do to keep `times` working under modification, they should in principle be correct

thht, 2018-10-23:
thanks @larsoner . i took a look and it looks like `decimate` uses `self._decim_slice` to tell the loading function what samples to discard but there seems to be no way to tell `_get_data()` to shift samples.

however, this all just works fine when the data is already loaded because then all one has to do (as far as i see it), it adjust `epochs._time` (i.e., `epochs.time` in the current master implementation.

so, what about adding a `shift_time` method that only works when data has already been loaded?

jasmainak, 2018-10-23:
There is already a [similar method](https://github.com/mne-tools/mne-python/blob/35f54e709e8d8e33d2c5996f37699ae7e4b1a651/mne/evoked.py#L267) in `evoked` -- you could probably copy parts of it

thht, 2018-10-23:
@jasmainak  yes, i am aware of that one. but `Evoked` does not have to deal lazy loading data...

jasmainak, 2018-10-23:
I would say it's fine to make it work only for `preload=True`

larsoner, 2018-10-23:
Let's do that for now and if we need it for `preload=False` we can always make it work later

PR #5651 (2018-10-25): MRG: Make epochs.times read only
Closes #5642.
closed 2018-10-25 by agramfort