mne-tools/mne-python #5643: BUG: Projection vectors

massich, 2018-10-22:
Some projections do not plot properly. When executing the following lines, the data of the first projection gets substituted by a vector of zeros (see figure)

```py
import mne
import matplotlib.pyplot as plt

FNAME= '/home/sik/Dropbox/mne-python/issues-pr-data/5641_proj_vectors/passive_raw.fif'

# getting some data ready
data_path = mne.datasets.sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'

# def test_xx():
raw = mne.io.read_raw_fif(FNAME, preload=True)
p1 = raw.info['projs'][0]
p1.plot_topomap()

raw_2 = mne.io.read_raw_fif(raw_fname, preload=True)
p2 = raw_2.info['projs'][0]
p2.plot_topomap()
```

[image]

Please find the data in [this dropbox folder](https://www.dropbox.com/sh/8e07r7a22rzwv34/AAC5pVEPfl7Q-9ccJPW6oUKZa?dl=0)

massich, 2018-10-22:
So far I found that the problem is here:
https://github.com/mne-tools/mne-python/blob/2636aa90eb364a41931bfd31b34a03e9fc042b0e/mne/viz/topomap.py#L310

at this point data still contains meaningful information:
```
-> data = _merge_grad_data(data[grad_pairs]).ravel()
(Pdb) data
array([ 0.        ,  0.        , -0.04986294,  0.        ,  0.        ,
       -0.05666957,  0.        ,  0.        , -0.03278815,  0.        ,
        0.        , -0.02590388,  0.        ,  0.        , -0.03766364,
        0.        ,  0.        , -0.05912703,  0.        ,  0.        ,
       -0.03077457,  0.        ,  0.        , -0.00951605,  0.        ,
        0.        , -0.08815657,  0.        ,  0.        , -0.08827741,
        0.        ,  0.        , -0.09489518,  0.        ,  0.        ,
       -0.06481268,  0.        ,  0.        , -0.07359767,  0.        ,
        0.        , -0.08506741,  0.        ,  0.        , -0.06199643,
        0.        ,  0.        , -0.04461724,  0.        ,  0.        ,
       -0.12610932,  0.        ,  0.        , -0.1444869 ,  0.        ,
        0.        , -0.14791939,  0.        ,  0.        , -0.11994962,
        0.        ,  0.        , -0.13436206,  0.        ,  0.        ,
       -0.11599123,  0.        ,  0.        , -0.09373209,  0.        ,
        0.        , -0.11744647,  0.        ,  0.        , -0.06848405,
        0.        ,  0.        , -0.07407323,  0.        ,  0.        ,
       -0.03095417,  0.        ,  0.        , -0.03290465,  0.        ,
        0.        , -0.14595643,  0.        ,  0.        , -0.15004401,
        0.        ,  0.        , -0.13788305,  0.        ,  0.        ,
       -0.1102547 ,  0.        ,  0.        , -0.10661658,  0.        ,
        0.        , -0.14208844,  0.        ,  0.        , -0.13833897,
        0.        ,  0.        , -0.13061485,  0.        ,  0.        ,
       -0.11547998,  0.        ,  0.        , -0.0954088 ,  0.        ,
        0.        , -0.08045314,  0.        ,  0.        , -0.06720123,
        0.        ,  0.        , -0.04089301,  0.        ,  0.        ,
       -0.05498577,  0.        ,  0.        , -0.07112252,  0.        ,
        0.        , -0.04767082,  0.        ,  0.        , -0.07041165,
        0.        ,  0.        , -0.09019454,  0.        ,  0.        ,
       -0.04386027,  0.        ,  0.        , -0.02447142,  0.        ,
        0.        ,  0.00242625,  0.        ,  0.        , -0.01714414,
        0.        ,  0.        , -0.03916721,  0.        ,  0.        ,
       -0.02713873,  0.        ,  0.        , -0.00493628,  0.        ,
        0.        , -0.01651539,  0.        ,  0.        , -0.00433761,
        0.        ,  0.        ,  0.04469855,  0.        ,  0.        ,
        0.06440534,  0.        ,  0.        ,  0.011707  ,  0.        ,
        0.        ,  0.034792  ,  0.        ,  0.        ,  0.01542465,
        0.        ,  0.        ,  0.06547109,  0.        ,  0.        ,
        0.0828331 ,  0.        ,  0.        ,  0.11579621,  0.        ,
        0.        ,  0.09541685,  0.        ,  0.        ,  0.15121306,
        0.        ,  0.        ,  0.16933501,  0.        ,  0.        ,
       -0.00485737,  0.        ,  0.        , -0.02292111,  0.        ,
        0.        ,  0.02747053,  0.        ,  0.        ,  0.05068428,
        0.        ,  0.        ,  0.12262075,  0.        ,  0.        ,
        0.1693447 ,  0.        ,  0.        ,  0.18730068,  0.        ,
        0.        ,  0.13583766,  0.        ,  0.        ,  0.08944295,
        0.        ,  0.        ,  0.0899028 ,  0.        ,  0.        ,
        0.14090514,  0.        ,  0.        ,  0.14135695,  0.        ,
        0.        ,  0.1783345 ,  0.        ,  0.        ,  0.20113003,
        0.        ,  0.        ,  0.19437158,  0.        ,  0.        ,
        0.19363311,  0.        ,  0.        , -0.01704232,  0.        ,
        0.        , -0.00223077,  0.        ,  0.        ,  0.05303189,
        0.        ,  0.        ,  0.02891105,  0.        ,  0.        ,
        0.11985461,  0.        ,  0.        ,  0.13667805,  0.        ,
        0.        ,  0.18172772,  0.        ,  0.        ,  0.1676001 ,
        0.        ,  0.        ,  0.02462523,  0.        ,  0.        ,
        0.04238194,  0.        ,  0.        ,  0.08429538,  0.        ,
        0.        ,  0.06791189,  0.        ,  0.        ,  0.14739263,
        0.        ,  0.        ,  0.09363393,  0.        ,  0.        ,
        0.12325267,  0.        ,  0.        ,  0.16960558,  0.        ,
        0.        ,  0.00810371,  0.        ,  0.        ,  0.03077618,
        0.        ,  0.        ,  0.07499995,  0.        ,  0.        ,
        0.04987418], dtype=float32)
(Pdb) grad_pairs
[0, 1, 3, 4, 6, 7, 9, 10, 12, 13, 15, 16, 18, 19, 21, 22, 24, 25, 27, 28, 30, 31, 33, 34, 36, 37, 39, 40, 42, 43, 45, 46, 48, 49, 51, 52, 54, 55, 57, 58, 60, 61, 63, 64, 66, 67, 69, 70, 72, 73, 75, 76, 78, 79, 81, 82, 84, 85, 87, 88, 90, 91, 93, 94, 96, 97, 99, 100, 102, 103, 105, 106, 108, 109, 111, 112, 114, 115, 117, 118, 120, 121, 123, 124, 126, 127, 129, 130, 132, 133, 135, 136, 138, 139, 141, 142, 144, 145, 147, 148, 150, 151, 153, 154, 156, 157, 159, 160, 162, 163, 165, 166, 168, 169, 171, 172, 174, 175, 177, 178, 180, 181, 183, 184, 186, 187, 189, 190, 192, 193, 195, 196, 198, 199, 201, 202, 204, 205, 207, 208, 210, 211, 213, 214, 216, 217, 219, 220, 222, 223, 225, 226, 228, 229, 231, 232, 234, 235, 237, 238, 240, 241, 243, 244, 246, 247, 249, 250, 252, 253, 255, 256, 258, 259, 261, 262, 264, 265, 267, 268, 270, 271, 273, 274, 276, 277, 279, 280, 282, 283, 285, 286, 288, 289, 291, 292, 294, 295, 297, 298, 300, 301, 303, 304]
(Pdb) 
```
However, `grad_pairs` takes all but the values that contains data.

what am I missing?

larsoner, 2018-10-22:
I don't understand as I responded to #5641, but the issue seems to be that this is a `mag` projector being shown with grad pairs. Setting `ch_type` should fix it, but this should be detectable.

massich, 2018-10-22:
(@larsoner yes that was in #5641 but for whatever reason, I could not even see my own comment nor any college in the lab. So I made an issue with the same content)

from @agramfort, changing this makes it work:
```diff
- p1.plot_topomap()
+ p1.plot_topomap(info=raw.copy().pick_types(meg='mag').info)
```
[image]

massich, 2018-10-22:
Similar thing happens here:
https://github.com/mne-tools/mne-python/blob/2636aa90eb364a41931bfd31b34a03e9fc042b0e/mne/viz/tests/test_topomap.py#L120

the resulting topomap is full blue, and the same thing happens. At the moment of calling `plot_proj_topomap` there's data in it. But when calling `plot_topomap(data, ..` data is an array of 0s.
```
(Pdb) data
array([-2.89551076e-02,  0.00000000e+00,  0.00000000e+00, -3.28691378e-02,
        0.00000000e+00,  0.00000000e+00, -1.54013634e-02,  0.00000000e+00,
        0.00000000e+00, -1.64652169e-02,  0.00000000e+00,  0.00000000e+00,
       -2.76559573e-02,  0.00000000e+00,  0.00000000e+00, -8.22870880e-02,
        0.00000000e+00,  0.00000000e+00, -6.31631166e-02,  0.00000000e+00,
        0.00000000e+00, -1.26191331e-02,  0.00000000e+00,  0.00000000e+00,
       -6.86531961e-02,  0.00000000e+00,  0.00000000e+00, -1.04377426e-01,
        0.00000000e+00,  0.00000000e+00, -1.46520004e-01,  0.00000000e+00,
        0.00000000e+00, -4.78750430e-02,  0.00000000e+00,  0.00000000e+00,
       -1.29438579e-01,  0.00000000e+00,  0.00000000e+00, -1.66978225e-01,
        0.00000000e+00,  0.00000000e+00, -1.51027441e-01,  0.00000000e+00,
        0.00000000e+00, -1.10901617e-01,  0.00000000e+00,  0.00000000e+00,
       -1.09113365e-01,  0.00000000e+00,  0.00000000e+00, -1.30161151e-01,
        0.00000000e+00,  0.00000000e+00, -1.65531546e-01,  0.00000000e+00,
        0.00000000e+00, -1.32578194e-01,  0.00000000e+00,  0.00000000e+00,
       -1.89775750e-01,  0.00000000e+00,  0.00000000e+00, -1.97292805e-01,
        0.00000000e+00,  0.00000000e+00, -1.86745733e-01,  0.00000000e+00,
        0.00000000e+00, -1.90683872e-01,  0.00000000e+00,  0.00000000e+00,
       -1.70296669e-01,  0.00000000e+00,  0.00000000e+00, -1.72050312e-01,
        0.00000000e+00,  0.00000000e+00, -1.41322568e-01,  0.00000000e+00,
        0.00000000e+00, -1.43627495e-01,  0.00000000e+00,  0.00000000e+00,
       -1.38891503e-01,  0.00000000e+00,  0.00000000e+00, -1.71277076e-01,
        0.00000000e+00,  0.00000000e+00, -1.28789082e-01,  0.00000000e+00,
        0.00000000e+00, -9.62548554e-02,  0.00000000e+00,  0.00000000e+00,
       -1.24820180e-01,  0.00000000e+00,  0.00000000e+00, -1.60948589e-01,
        0.00000000e+00,  0.00000000e+00, -1.95158675e-01,  0.00000000e+00,
        0.00000000e+00, -1.85143143e-01,  0.00000000e+00,  0.00000000e+00,
       -1.85745820e-01,  0.00000000e+00,  0.00000000e+00, -1.85687110e-01,
        0.00000000e+00,  0.00000000e+00, -1.59779191e-01,  0.00000000e+00,
        0.00000000e+00, -1.22068413e-01,  0.00000000e+00,  0.00000000e+00,
       -1.04489453e-01,  0.00000000e+00,  0.00000000e+00, -1.43614948e-01,
        0.00000000e+00,  0.00000000e+00, -6.05320744e-02,  0.00000000e+00,
        0.00000000e+00, -3.68443541e-02,  0.00000000e+00,  0.00000000e+00,
       -9.16534364e-02,  0.00000000e+00,  0.00000000e+00, -1.42298177e-01,
        0.00000000e+00,  0.00000000e+00, -6.85598403e-02,  0.00000000e+00,
        0.00000000e+00, -1.77562460e-02,  0.00000000e+00,  0.00000000e+00,
        1.75734493e-03,  0.00000000e+00,  0.00000000e+00, -5.08735180e-02,
        0.00000000e+00,  0.00000000e+00, -1.93720069e-02,  0.00000000e+00,
        0.00000000e+00, -1.42914737e-02,  0.00000000e+00,  0.00000000e+00,
        6.27118061e-05,  0.00000000e+00,  0.00000000e+00, -2.66114599e-03,
        0.00000000e+00,  0.00000000e+00,  3.18721915e-03,  0.00000000e+00,
        0.00000000e+00,  2.93999724e-02,  0.00000000e+00,  0.00000000e+00,
        3.15912142e-02,  0.00000000e+00,  0.00000000e+00,  6.06454117e-03,
        0.00000000e+00,  0.00000000e+00,  1.06812138e-02,  0.00000000e+00,
        0.00000000e+00, -3.50147411e-02,  0.00000000e+00,  0.00000000e+00,
       -1.08254217e-02,  0.00000000e+00,  0.00000000e+00,  3.25581282e-02,
        0.00000000e+00,  0.00000000e+00,  5.44461682e-02,  0.00000000e+00,
        0.00000000e+00,  5.17732948e-02,  0.00000000e+00,  0.00000000e+00,
        7.78989419e-02,  0.00000000e+00,  0.00000000e+00,  7.51223937e-02,
        0.00000000e+00,  0.00000000e+00, -8.22032541e-02,  0.00000000e+00,
        0.00000000e+00, -1.22374713e-01,  0.00000000e+00,  0.00000000e+00,
       -9.40561965e-02,  0.00000000e+00,  0.00000000e+00, -5.44069111e-02,
        0.00000000e+00,  0.00000000e+00,  1.46951070e-02,  0.00000000e+00,
        0.00000000e+00,  5.87537326e-02,  0.00000000e+00,  0.00000000e+00,
        8.91550779e-02,  0.00000000e+00,  0.00000000e+00,  5.05104810e-02,
        0.00000000e+00,  0.00000000e+00, -3.31369080e-02,  0.00000000e+00,
        0.00000000e+00, -3.27853151e-02,  0.00000000e+00,  0.00000000e+00,
        2.07648147e-02,  0.00000000e+00,  0.00000000e+00,  1.88973453e-02,
        0.00000000e+00,  0.00000000e+00,  6.13227300e-02,  0.00000000e+00,
        0.00000000e+00,  9.76517051e-02,  0.00000000e+00,  0.00000000e+00,
        8.74372199e-02,  0.00000000e+00,  0.00000000e+00,  8.47573429e-02,
        0.00000000e+00,  0.00000000e+00, -1.15081370e-01,  0.00000000e+00,
        0.00000000e+00, -7.75375888e-02,  0.00000000e+00,  0.00000000e+00,
       -4.82228100e-02,  0.00000000e+00,  0.00000000e+00, -9.21199620e-02,
[... 1 lines cut ...]
        0.00000000e+00,  5.99749051e-02,  0.00000000e+00,  0.00000000e+00,
        9.18874145e-02,  0.00000000e+00,  0.00000000e+00,  6.49671927e-02,
        0.00000000e+00,  0.00000000e+00, -2.27321368e-02,  0.00000000e+00,
        0.00000000e+00,  2.22607199e-02,  0.00000000e+00,  0.00000000e+00,
[... cut ...]
```
```
(Pdb) data
array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.],
      dtype=float32)
(Pdb) 
```

larsoner, 2018-11-07:
Fixed by #5641

closed 2018-11-07 by larsoner