mne-tools/mne-python #5645: CI badges in readme out of sync

QB3, 2018-10-23:
#### Describe the bug

in the github main page (see screenshot) the badges show all the CI failing. 

[image]

checking if thats the case (by clicking on it) Travis is all green. see this:

[image]

It seems to mean that the badges are not synchronized.

#### Expected results
Since the tests are passed I would expect the badges to be green.

#### Actual results
The bagdes are red, and it states "failing".

agramfort, 2018-10-23:
well our current master is actually broken :(

but the svg badges should point to the master branch as done here:

https://docs.travis-ci.com/user/status-images/

larsoner, 2018-10-23:
It appears correct see

https://travis-ci.org/mne-tools/mne-python/jobs/444753486

closed 2018-10-23 by larsoner

agramfort, 2018-10-23:
hum :(

but we still should fix the badges to point to master branch or maybe it's
not necessary?

larsoner, 2018-10-23:
in README.rst:
```
.. |Travis| image:: https://api.travis-ci.org/mne-tools/mne-python.png?branch=master
```
so it should already