mne-tools/mne-python #5650: Unable to add source for libXp6 package

nordme, 2018-10-23:
Instructions for resolving the missing libXp.so.6 package issue were helpfully added to the MNE-C documentation: http://mne-tools.github.io/dev/install_mne_c.html#troubleshooting. Unfortunately, with my Ubuntu 18.04 setup, the following error pops up when attempting the first command:

 `W: GPG error: http://security.ubuntu.com/ubuntu precise-security InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 40976EAF437D05B5`

`E: The repository 'http://security.ubuntu.com/ubuntu precise-security InRelease' is not signed.`

`N: Updating from such a repository can't be done securely, and is therefore disabled by default.`

`N: See apt-secure(8) manpage for repository creation and user configuration details.`

Attached is the full traceback. 
[image]

nordme, 2018-10-23:
Thanks, Joan! Thinking of mne documentation, what is the best place to put these GPG key instructions do you think? From a user perspective, it would be fantastic if a couple lines of code to obtain the keys could be added to the troubleshooting snippet installing the libXp package.

massich, 2018-10-23:
Did it work?

I think that the instructions should be added here. But they change
overtime. So the snipped might be outdated pretty soon. But it might be
worth it. Maybe when it stop working someone would rase the issue.

Would you like to wite a PR for it?

nordme, 2018-10-25:
@massich Hmm. Getting the key for the NeuroDebian repo did not work for me. The initial command `wget -O- http://neuro.debian.net/lists/bionic.us-nh.libre | sudo tee /etc/apt/sources.list.d/neurodebian.sources.list` results in my terminal sitting around doing nothing. Take a look. This is how the terminal has looked for the past hour.
[image]

larsoner, 2018-10-25:
It seems like the critical command will be the second one (the `sudo apt-key` one), can you try that then see if the `libXp` install works?

larsoner, 2018-10-25:
...actually just try:
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 40976EAF437D05B5
```

nordme, 2018-10-25:
Hi,

I tried the second command. And I tried to add the key from a different
keyserver as well (keyserver.ubuntu.com). None of the keyservers I tried
seems to have the key for the repository in the proposed fix.

Erica Peterson
Research Scientist
Institute for Learning and Brain Sciences
University of Washington
Box 357988
Seattle, WA 98195
(206) 685-2842

larsoner, 2018-10-25:
Okay how about these two lines:
```
sudo apt -o Acquire::AllowInsecureRepositories=true -o Acquire::AllowDowngradeToInsecureRepositories=true update
sudo apt -o APT::Get::AllowUnauthenticated=true install libxp6
```

nordme, 2018-10-25:
To specify, I did try the key command with

40976EAF437D05B5

as the key number.

Erica Peterson
Research Scientist
Institute for Learning and Brain Sciences
University of Washington
Box 357988
Seattle, WA 98195
(206) 685-2842

nordme, 2018-10-25:
Eric's lines worked great.

PR #5667 (2018-10-26): DOC: Update commands
Closes #5650
closed 2018-10-26 by massich