mne-tools/mne-python #5661: Read montage that isn't in built-in list?

shateeshb, 2018-10-24:
Hi all, I am a CS student who is new to working with EEG, so please bear with me. I am using EEG data that was collected using the ANT Duke Waveguard, with 128 channels. Is there any way to import the montage file (.txt), since mne.channels.read_montage() won't work?

cbrnr, 2018-10-25:
Assuming you have your montage file in a suitable format that we can read and you have it in the current working directory (in the following example it is called `test_montage.txt`), you can do:

```
from os.path import abspath
montage = mne.channels.read_montage(abspath("test_montage.txt"))
```

shateeshb, 2018-10-25:
> Assuming you have your montage file in a suitable format that we can read and you have it in the current working directory (in the following example it is called `test_montage.txt`), you can do:
> 
> ```
> from os.path import abspath
> montage = mne.channels.read_montage(abspath("test_montage.txt"))
> ```

I tried using just the path to the file, but I was given an error that the parameter 'kind' was not specified. 

When I did specify the type (using the naming convention outlined in the documentation, where the kind = the file name without the file extension), I got this error: 
[image]

here is the montage file that I was using: 

[ANT_DukeWaveGuard_128_electrode_montage_copy.txt](https://github.com/mne-tools/mne-python/files/2515235/ANT_DukeWaveGuard_128_electrode_montage_copy.txt)

Does naming convention matter in this case?

agramfort, 2018-10-25:
we don't have a parser to read ANT files. Your file is not a plain txt file
so the extension should match the extension of an ANT montage file.

I don't have experience with this

cbrnr, 2018-10-25:
@mmagnuski this is a bit weird, it doesn't work unless I specify the absolute path. Also, I just pass the full path to the first argument and I don't use the `path` argument. Not sure why there is this argument in the first place.

@agramfort this is a .txt file, but I don't know what kind of format we support. @shateeshb maybe you can have a look at the existing montage files and try to convert yours?

shateeshb, 2018-10-26:
> @mmagnuski this is a bit weird, it doesn't work unless I specify the absolute path. Also, I just pass the full path to the first argument and I don't use the `path` argument. Not sure why there is [...]

okay I will try converting the montage file. Thanks for the help!

massich, 2018-10-26:
Aren't the supported montage [here](https://github.com/mne-tools/mne-python/blob/0db3c5776de04ec2f926295f4a191010efe49c39/mne/channels/montage.py#L169)

@shateeshb could you draft a PR with some example data to support your format?

Complete newbie on this topic but is this montage related to `dig_montage` i.e: #5615, #5617 ??

agramfort, 2018-10-26:
there is a previous issue on the support of ANT format :
https://github.com/mne-tools/mne-python/issues/3609

so far it's stalled

shateeshb, 2018-11-01:
> Aren't the supported montage [here](https://github.com/mne-tools/mne-python/blob/0db3c5776de04ec2f926295f4a191010efe49c39/mne/channels/montage.py#L169)
>
> @shateeshb could you draft a PR with some [...]

Apologies for the late reply, I had to wait for permission from the original researcher who collected the data if  I could share a sample of it. 
[sample-ANT-data.zip](https://github.com/mne-tools/mne-python/files/2539868/sample-ANT-data.zip)

Here is a sample of the data, it consists of a .set, .evt and .trg file of one recording of a trial using the ANT Duke Waveguard cap.

agramfort, 2018-11-02:
thanks

can you give a try to contributing python code to read these files?

shateeshb, 2018-11-05:
> thanks can you give a try to contributing python code to read these files?

I managed to get mne.channels.read_montage() to work on the montage file that I originally asked about. I had to manually clean up the txt file, but it works. Thanks for the help

closed 2018-11-05 by shateeshb